wmbusmeters/wmbusmeters #1475: Qundis QHeat 5.5

Heavendenied81, 2025-01-24:
### Type of request

New device support

### Meter description

HI. I'm really new to this so sorry if maybe my information is not enough or wrong...

I tried to get my HomeAssistant work with my new Qundis qHeat 5.5 with integrated wirless mBus.
They are new and not configured from any company before.
I found out I need to use C1 protokoll and then get this in the Logs:

(cul) warning: the hex string is not proper! Ignoring telegram!
Received telegram from: 68167882
          manufacturer: (QDS) Qundis, Germany (0x4493)
                  type: Radio converter (meter side) (0x37)
                   ver: 0x47
                device: cul
                  rssi: -43 dBm
                driver: unknown!
telegram=|_5844934482781668473778077982781668934447040DFF5F3500828D00358754C3875D02AD44CD324DCA2661C8DC45D9D39C4D11690B5A1FEB03879993EC470517019411280CD9E19D9EB598857402FD170000046D220A3831|+33

Can anyone help me geeting further with this?

Thanks in advance!

### Logged telegrams from meter in decrypted format

```shell
|_5844934482781668473778077982781668934447040DFF5F3500828D00358754C3875D02AD44CD324DCA2661C8DC45D9D39C4D11690B5A1FEB03879993EC470517019411280CD9E19D9EB598857402FD170000046D220A3831|+33
|5844934482781668473778_077982781668934447040DFF5F3500828E0035797A29E8638A74D5261C5D1B726E7AEBAC1F34D2FD4325E6DDA6423990BD53FCA2DDF9DB8048B190C0D943C37AC3AC0A02FD170000046D270A3831|+210
telegram=|5844934482781668473778_077982781668934447040DFF5F3500828E003565A0E0277B1C8CD599226FEBF9D24FE0AA120CE542FE8EF466948F05FB4056CBE09A78E9B267E04F7780FFD5C22E51AB02FD170000046D290A3831|+319
telegram=|5844934482781668473778_077982781668934447040DFF5F3500828E00352C3F06D074D574C4FA86F0601303E2A889B5D5C94DBB647F668902A8F22C6FF33ACA64E82BE21222E201602231C0A8EA02FD170000046D2B0A3831|+433
telegram=|5844934482781668473778_077982781668934447040DFF5F3500828F0035306F6952B1A522482A9141C0CE6F87CBB591BCAD878FFB3035A93B84A632DD9C54143EB1F9EB49AFA376BDC2DACE9CD202FD170000046D2E0A3831|+646
telegram=|5844934482781668473778_077982781668934447040DFF5F3500828F00357D8F8806C6981182DA2FC7F082270C3744209DEDDD3D163D1B535B182B1A7C2BA13A54769374C5A2FAA3799D31925F1602FD170000046D320A3831|+870
telegram=|5844934482781668473778_077982781668934447040DFF5F3500829000357D86B735AC29FB7FB52C58995F41125B6CB44DD48DBF1E097404FAEFE3F744F92DA828C2C1E871935AEC9962BED5036202FD170000046D360A3831|+1087
```

gotsila, 2025-02-02:
Hello,

can you found a string like this:  M=QDS.V=46.T=04 

How have you integrate it? As Add-On in HA?

Sounds like the same problem as here:     https://github.com/wmbusmeters/wmbusmeters/issues/1347

schmyd, 2025-02-27:
This issue has been reported in multiple threads. I've provided a home assistant workaround in #1497

closed 2025-05-29 by github-actions[bot]