wmbusmeters/wmbusmeters #1478: No Data

Kasper027, 2025-01-28:
### Type of request

Question

### HAOS version

14.1

### wmbusmeters version

1.18.0-RC1

### Your message goes here

Hello. I'm trying to read data from the M-Bus meters. I have wired meters. I use the M-Bus micro-master from Relay. I've already had the whole thing running with a separate Raspberry and the system from The78Mole. Unfortunately, the Raspberry is defective (raspberry hardware and flash). Now I've seen this project and would like to run it all in the Home Assistant. Who can help me?

I have the following log output:

```
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun flask (no readiness notification)
services-up: info: copying legacy longrun nginx (no readiness notification)
services-up: info: copying legacy longrun wmbusmeters (no readiness notification)
[12:30:48] INFO: Starting wmbusmeters service.
s6-rc: info: service legacy-services successfully started
[12:30:48] INFO: Starting web configuration service.
[12:30:49] INFO: CONFIG_CONF ...
[12:30:49] INFO: {"device":"auto:t1","donotprobe":"/dev/ttyAMA0","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":false,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[12:30:49] INFO: CONFIG_METERS ...
[12:30:49] INFO: []
[12:30:49] INFO: Syncing wmbusmeters configuration ...
[12:30:49] INFO: Registering meters ...
[12:30:49] INFO: Generating MQTT configuration ... 
[12:30:49] INFO: Starting Nginx...
[12:30:49] INFO: Broker core-mosquitto will be used.
2025/01/28 12:30:49 [notice] 91#91: using the "epoll" event method
2025/01/28 12:30:49 [notice] 91#91: nginx/1.24.0
2025/01/28 12:30:49 [notice] 91#91: OS: Linux 6.6.66-haos
2025/01/28 12:30:49 [notice] 91#91: getrlimit(RLIMIT_NOFILE): 1073741816:1073741816
2025/01/28 12:30:49 [notice] 91#91: start worker processes
2025/01/28 12:30:49 [notice] 91#91: start worker process 201
[12:30:49] INFO: 
MQTT Discovery ...
[12:30:49] INFO: MQTT Discovery cleanup...
[12:30:50] INFO: Running wmbusmeters ...
No wmbus device detected, waiting for a device to be plugged in.
No meters configured. Printing id:s of all telegrams heard!
```

I just don't understand how I could get the data then. Do I need to use different configurations for wired devices?

Kasper027, 2025-01-28:
This data is displayed under Hardware in HomeAssistant for the adapter:

```
ID:
/dev/serial/by-id/usb-Silicon_Labs_Relay_M-Bus_Micro-Master_USB_7302-if00-port0
Attribute:
DEVLINKS: >-
  /dev/serial/by-id/usb-Silicon_Labs_Relay_M-Bus_Micro-Master_USB_7302-if00-port0
  /dev/serial/by-path/pci-0000:02:1b.0-usb-0:3:1.0-port0
  /dev/serial/by-path/pci-0000:02:1b.0-usbv2-0:3:1.0-port0
DEVNAME: /dev/ttyUSB2
DEVPATH: >-
  /devices/pci0000:00/0000:00:1e.0/0000:02:1b.0/usb2/2-3/2-3:1.0/ttyUSB2/tty/ttyUSB2
ID_BUS: usb
ID_MODEL: Relay_M-Bus_Micro-Master_USB
ID_MODEL_ENC: Relay\x20M-Bus\x20Micro-Master\x20USB
ID_MODEL_ID: ea60
ID_PATH: pci-0000:02:1b.0-usb-0:3:1.0
ID_PATH_TAG: pci-0000_02_1b_0-usb-0_3_1_0
ID_PATH_WITH_USB_REVISION: pci-0000:02:1b.0-usbv2-0:3:1.0
ID_REVISION: '0100'
ID_SERIAL: Silicon_Labs_Relay_M-Bus_Micro-Master_USB_7302
ID_SERIAL_SHORT: '7302'
ID_TYPE: generic
ID_USB_DRIVER: cp210x
ID_USB_INTERFACES: ':ff0000:'
ID_USB_INTERFACE_NUM: '00'
ID_USB_MODEL: Relay_M-Bus_Micro-Master_USB
ID_USB_MODEL_ENC: Relay\x20M-Bus\x20Micro-Master\x20USB
ID_USB_MODEL_ID: ea60
ID_USB_REVISION: '0100'
ID_USB_SERIAL: Silicon_Labs_Relay_M-Bus_Micro-Master_USB_7302
ID_USB_SERIAL_SHORT: '7302'
ID_USB_TYPE: generic
ID_USB_VENDOR: Silicon_Labs
ID_USB_VENDOR_ENC: Silicon\x20Labs
ID_USB_VENDOR_ID: 10c4
ID_VENDOR: Silicon_Labs
ID_VENDOR_ENC: Silicon\x20Labs
ID_VENDOR_ID: 10c4
MAJOR: '188'
MINOR: '2'
SUBSYSTEM: tty
TAGS: ':systemd:'
USEC_INITIALIZED: '13964844249'
```

Kasper027, 2025-02-03:
Hello. Does anyone have an idea

kzajac83, 2025-02-19:
Not using HA (due to problem HW M-Bus mater and HA server) however in linux I using this command:

```
wmbusmeters --analyze --oneshot --pollinterval=600s --format=json --ppjson /dev/ttyUSB0:mbus:2400 licznik /etc/wmbusmeters.drivers.d/aptmbusna.xmq:mbus p123 NOKEY

wmbusmeters --analyze --oneshot --pollinterval=600s --format=json --ppjson /dev/ttyS0:mbus:2400 licznik elf2:mbus 1245678 NOKEY
```

and other programs for testing:

```
mbus-serial-scan 
mbus-serial-select-secondary

```

My config is simple:

```
root@wyse:~# cat /etc/wmbusmeters.conf
#loglevel=debug
loglevel=normal
# Remember to change auto here to the device you are going to use in production.
device=SERIAL_1=/dev/ttyUSB0:mbus:2400
device=SERIAL_2=/dev/ttyS0:mbus:2400
```

And some meters:

```
root@wyse:~# cat /etc/wmbusmeters.d/some_file_name
name=MBUS_METER
id=12345678
driver=kamheat:SERIAL_1:mbus
pollinterval=180s
calculate_total_energy_gj=total_energy_consumption_kwh
#identitymode=full
#metershell=/mqtt_discover.sh
```

Read this: https://github.com/wmbusmeters/wmbusmeters/issues/435

Kasper027, 2025-02-25:
Hey @kzajac83 thank you for your answer. With your code I was able to read out my current meter reading. Now I have to see if I can put together a driver myself. But I am already getting data. That is great. Thank you again for that

closed 2025-02-25 by Kasper027