wmbusmeters/wmbusmeters #1484: kamheat: MULTICAL 403 (MC403) doesn't work - compressed and signature unknown

exetico, 2025-02-05:
I have already used the util to decrypt data from my `multical21`, which does work as expected. Thank you, for this great solution!

Now I'm ready to get my Multical 403 to work, too.  

But I'm facing a wierd problem. At least, that's what I think?

Here's a telegram from the Multical 403:

```
3F442D2C9534477134048D20375A64FD224054B02BCC391E9E87C7E177960D24AA60C0D15E13B3DC23AB541A98BF5A40357105FAE61A865FACE2A89E1D845005 (64) C1 B
```

It does match the correct serial number, **however**, it's not able to extract the data.

I've provided the decryption key in the webpage, but it doesn't work.

I have the signture details, as part of my details related to the decryption key (DEK), but I don't want to share it here.

What can I do, 

```
Auto driver    : kamheat
Similar driver : unknown 00/00
Using driver   : kamheat 00/00
000   : 3f length (63 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 2d2c dll-mfct (KAM)
004   : 95344771 dll-id (71473495)
008   : 34 dll-version
009   : 04 dll-type (Heat meter)
010   : 8d ell-ci-field (ELL: Extended Link Layer II (8 Byte))
011   : 20 ell-cc (slow_resp sync)
012   : 36 ell-acc
013   : 3564fd22 sn (AES_CTR)
017   : c1c3 payload crc (calculated c1c3 OK)
019   : 79 tpl-ci-field (EN 13757-3 Application Layer with Compact frame (no tplh))
020   : 292e format signature (unknown)
022 CC: E1873FF1000015BA010060E90000F7AD02003119180C58000000230000009C0000001ACC000000000000 compressed and signature unknown

{
    "_":"telegram",
    "media":"heat",
    "meter":"kamheat",
    "name":"",
    "id":"71473495",
    "timestamp":"1970-01-01T00:00:00Z"
}

Using: wmbusmeters: 1.17.1-70-g41a2b48
41a2b4808f00367af7bbaac57e221539825babbb
```

I've looked at https://github.com/wmbusmeters/wmbusmeters/pull/147, but I'm not really sure if that's related.

exetico, 2025-02-05:
While adding more details to the issue, I noticed that I've received **one** telegram with a longer payload.

```
[19:59:05][I][wmbus:104]: kamheat [0x71473495] RSSI: -53dBm T: 3F442D2C9534477134048D2033EA63FD22E4774CC10E74EDCBB17A5B70682A9482DF6C34456E131E430B124494994E8803348C90375346E09F1205E74061AAA9 (64) C1 B
[20:00:42][I][wmbus:104]: kamheat [0x71473495] RSSI: -53dBm T: 3F442D2C9534477134048D20340F64FD222AD8CF00A2E28BFB78E4538F51B8D284A53191A164F389DD4E0F70E20F5FBB7BD6BA2861874A785685403AC9287E11 (64) C1 B
[20:03:55][I][wmbus:104]: kamheat [0x71473495] RSSI: -53dBm T: 3F442D2C9534477134048D20363564FD228ECDDCBA366D8D12AF4F126C17CE9E7ACF1A5C54891CBBE4EB511CC7CCBC35B1FD5EE5786C5BFBA882055C3D48A110 (64) C1 B
[20:05:32][I][wmbus:104]: kamheat [0x71473495] RSSI: -53dBm T: 3F442D2C9534477134048D20375A64FD224054B02BCC391E9E87C7E177960D24AA60C0D15E13B3DC23AB541A98BF5A40357105FAE61A865FACE2A89E1D845005 (64) C1 B
[20:11:57][I][wmbus:104]: kamheat [0x71473495] RSSI: -53dBm T: 3F442D2C9534477134048D203BBA64FD22B8775C10282213A7A3FFDF2DED1805CA748F5D4FAB70FDA4E2DC5E067F619CC990E96866489CB26592AA63F7638F16 (64) C1 B
[20:13:33][I][wmbus:104]: kamheat [0x71473495] RSSI: -53dBm T: 3F442D2C9534477134048D203CDF64FD22A3A4EA4A20AF98E1D81070B795D6F977AAB8F357270278B287697B5906C9AF2ACBED7D894357D43AA56F3926F91AB0 (64) C1 B
[20:26:21][I][wmbus:104]: kamheat [0x71473495] RSSI: -53dBm T: 3F442D2C9534477134048D2044AF65FD22F971EECF9184A5D4A41D9D0DBA41A18F82CED315FB14DB8434C0FF23380B7C8010D02B93247DB60E748E4BFDA5642B (64) C1 B
[20:27:18][I][wmbus:104]: kamheat [0x71473495] RSSI: -53dBm T: 09472D2C953447713404 (10) C1 A
[20:31:09][I][wmbus:104]: kamheat [0x71473495] RSSI: -53dBm T: 3F442D2C9534477134048D2047FA65FD220A0DAFA35AFFBF0BA2528CB5B6579E2A9A70801FA382C9AADAB7E31A7A11DEADF0DD71A670F7953F05C4D797C64200 (64) C1 B
[20:34:20][I][wmbus:104]: kamheat [0x71473495] RSSI: -53dBm T: 54442D2C9534477134048D20492066FD2259284768C85528149BD24C8A6D43E0BDBC3CEB75A6988487886C5AC5822C77DFDCF721C4C4F58769280F214717F3930021CA0F91ED7E1FED78A1364BB344C1CBD2C951C7 (85) C1 B
[20:35:56][I][wmbus:104]: kamheat [0x71473495] RSSI: -53dBm T: 3F442D2C9534477134048D204A3566FD22983AC099A75D3B27CD7CB7E48E9B125BFEBBE15FD73295B5C2C3F5F150AE57F120597FDE30B9965496190CBCF71CC8 (64) C1 B
```

That one (`54442D2C9534477134048D20492066FD2259284768C85528149BD24C8A6D43E0BDBC3CEB75A6988487886C5AC5822C77DFDCF721C4C4F58769280F214717F3930021CA0F91ED7E1FED78A1364BB344C1CBD2C951C7` worked as expected, and encrypted like it should.

But now I'm wondering if the other payload does contain anything interesting? Or?

Uffe81, 2025-03-07:
I have the exact same issue with my Multical 303. I get telegrams ca every 96s. Kamstrup says I can change to 15 s but it requires proprietary hardware to do this (optical eye). So basically I have data every ca 15 minutes from the multical. All the short frames are thrown out and wmbusmeters does not "learn" how to decode them when it receives a long telegram. 

I'm using the esp-32 component from here (version 3.4.3) https://github.com/SzczepanLeon/esphome-components/blob/main/docs/version_3.md

closed 2025-07-10 by github-actions[bot]