wmbusmeters/wmbusmeters #1491: Bunch of CRC errors

mStirner, 2025-02-11:
Hey, i just started using wmbusmeters with docker.
It seems like it received the first telegram correctly, and then just prints CRC errors.
Do i have to configure something or why are there only CRC errors?

```
(rtlwmbus) using command: ERRFILE=$(mktemp -t wmbusmeters_rtlsdr.XXXXXXXXXXXX) ; echo ERRFILE=$ERRFILE ;  date -Iseconds > $ERRFILE ; tail -f $ERRFILE & /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>>$ERRFILE | /usr/bin/rtl_wmbus -s -f
(rtlwmbus) opening 31294080
(serial) EVENT thread interrupted
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "ERRFILE=$(mktemp -t wmbusmeters_rtlsdr.XXXXXXXXXXXX) ; echo ERRFILE=$ERRFILE ;  date -Iseconds > $ERRFILE ; tail -f $ERRFILE & /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>>$ERRFILE | /usr/bin/rtl_wmbus -s -f"
(serialcmd) opened /bin/sh pid 19 fd 6 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) received ascii "ERRFILE=/tmp/wmbusmeters_rtlsdr.XXXXXXPmOEPh<0A>rtl_wmbus: monitoring flow<0A>2025-02-11T11:13:27+01:00<0A>"
(rtlwmbus) checkRTLWMBusFrame "ERRFILE=/tmp/wmbusmeters_rtlsdr.XXXXXXPmOEPh<0A>rtl_wmbus: monitoring flow<0A>2025-02-11T11:13:27+01:00<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "rtl_wmbus: monitoring flow<0A>2025-02-11T11:13:27+01:00<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "2025-02-11T11:13:27+01:00<0A>"
(rtlwmbus) only text
(serial) received ascii "Found 1 device(s):<0A>  0:  Nooelec, NESDR SMArt v5, SN: 31294080<0A><0A>Using device 0: Generic RTL2832U OEM<0A>Detached kernel driver<0A>Found Rafael Micro R820T tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) checkRTLWMBusFrame "Found 1 device(s):<0A>  0:  Nooelec, NESDR SMArt v5, SN: 31294080<0A><0A>Using device 0: Generic RTL2832U OEM<0A>Detached kernel driver<0A>Found Rafael Micro R820T tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "  0:  Nooelec, NESDR SMArt v5, SN: 31294080<0A><0A>Using device 0: Generic RTL2832U OEM<0A>Detached kernel driver<0A>Found Rafael Micro R820T tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "<0A>Using device 0: Generic RTL2832U OEM<0A>Detached kernel driver<0A>Found Rafael Micro R820T tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Using device 0: Generic RTL2832U OEM<0A>Detached kernel driver<0A>Found Rafael Micro R820T tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Detached kernel driver<0A>Found Rafael Micro R820T tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Found Rafael Micro R820T tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Reading samples in async mode...<0A>"
(rtlwmbus) only text
(serial) received ascii "T1;1;1;2025-02-11 11:14:44.918773;27;29;41566006;0x32446850066056416980a0119f310800b024000059074e07000000010500000000000000000000000000000000000000000002<0A>T1;1;1;2025-02-11 11:14:44.918881;31;31;41566006;0x32446850066056416980a0119f310800b024000059074e07000000010500000000000000000000000000000000000000000002<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;1;1;2025-02-11 11:14:44.918773;27;29;41566006;0x32446850066056416980a0119f310800b024000059074e07000000010500000000000000000000000000000000000000000002<0A>T1;1;1;2025-02-11 11:14:44.918881;31;31;41566006;0x32446850066056416980a0119f310800b024000059074e07000000010500000000000000000000000000000000000000000002<0A>"
(rtlwmbus) received full frame
(meter) no meter handled  checking 0 templates.
(wmbus) parseDLL @0 51
(wmbus) parseELL @10 41
(wmbus) parseNWL @10 41
(wmbus) parseAFL @10 41
(wmbus) parseTPL @10 41
(wmbus) parseDLL @0 51
(telegram) DLL L=32 C=44 (from meter SND_NR) M=5068 (TCH) A=41566006 VER=69 TYPE=80 (Heat Cost Allocator) (driver fhkvdataiii) DEV=rtlwmbus[31294080] RSSI=27
(wmbus) parseELL @10 41
(wmbus) parseNWL @10 41
(wmbus) parseAFL @10 41
(wmbus) parseTPL @10 41
(telegram) TPL CI=a0
Received telegram from: 41566006
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat Cost Allocator (0x80)
                   ver: 0x69
                device: rtlwmbus[31294080]
                  rssi: 27 dBm
                driver: fhkvdataiii
[... 143 lines cut ...]
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;0;0;2025-02-11 11:24:07.546683;31;28;41FF48FF;0x3244ffffff48ff4169ffa0119f319b08b0246f0316ff55ffffff01ada0ff59ffff00ff0000ff4000ff00ffff09ff58ff55ff7e<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2025-02-11 11:24:07.546683;31;28;41FF48FF;0x3244ffffff48ff4169ffa0119f319b08b0246f0316ff55ffffff01ada0ff59ffff00ff0000ff4000ff00ffff09ff58ff55ff7e<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;0;0;2025-02-11 11:25:42.483366;28;24;FF86FFFF;0x32ffff52ffff86ff69ffa011ffffff02b0ffcfffffffff073bffffffffff3bffffffffffff00ffffffff004cff04ffffffff35<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2025-02-11 11:25:42.483366;28;24;FF86FFFF;0x32ffff52ffff86ff69ffa011ffffff02b0ffcfffffffff073bffffffffff3bffffffffffff00ffffffff004cff04ffffffff35<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;0;0;2025-02-11 11:26:52.199158;51;32;4132FF77;0x32ffff5077ff3241ff80a0119fffffffffff6fffff0bff0e719801adffff59ff00000000ffff00ffff00ffffff28ff5d55847e<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2025-02-11 11:26:52.199158;51;32;4132FF77;0x32ffff5077ff3241ff80a0119fffffffffff6fffff0bff0e719801adffff59ff00000000ffff00ffff00ffffff28ff5d55847e<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;0;0;2025-02-11 11:27:30.209402;22;31;FFFFFFFF;0x32ff08ffffffffffffffff65ffff2fffffffffffffffffffffffffffffe1ffffffffffffd6ffffffffffeeff0fffsince rtl_wmbus reports that CRC checks failed1923ffffff<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2025-02-11 11:27:30.209402;22;31;FFFFFFFF;0x32ff08ffffffffffffffff65ffff2fffffffffffffffffffffffffffffe1ffffffffffffd6ffffffffffeeff0fff1923ffffff<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
(serial) received ascii "T1;0;0;2025-02-11 11:28:33.949427;20;35;FF324877;0x32446850774832ff6980ff119f319b08b0246f03070b6d0e72ff01ffa0ffffff000000ff0000ff00ffff00ff09ff58ff55ff7e<0A>"
(rtlwmbus) checkRTLWMBusFrame "T1;0;0;2025-02-11 11:28:33.949427;20;35;FF324877;0x32446850774832ff6980ff119f319b08b0246f03070b6d0e72ff01ffa0ffffff000000ff0000ff00ffff00ff09ff58ff55ff7e<0A>"
(rtlwmbus) telegram received but incomplete or with errors, since rtl_wmbus reports that CRC checks failed.
(rtlwmbus) error in received message.
```

Thanks in advance.

mStirner, 2025-02-11:
Could it be because my meters may not be supported?
I have Brunata/Metrona devices:

[image]
[image]

kzajac83, 2025-02-16:
3 potential options:

1. Try find firmware for your SDR and upgrade it 
2. Maybe similar case like nanoCUL (HW issue)?
3. Or just a lot interferences in the air

Do you read correct any telegram?

mStirner, 2025-02-17:
> Do you read correct any telegram?

No, thats the complete output after plugging the SDR in and start the docker container.

I use the NOELEC 2A8GD-SMArt-v5 SDR stick.

kzajac83, 2025-02-19:
You are read one frame correctly:

https://wmbusmeters.org/analyze/32446850066056416980a0119f310800b024000059074e07000000010500000000000000000000000000000000000000000002

However please check driver file, on bottome you can found example telegrams:

https://github.com/wmbusmeters/wmbusmeters/blob/master/src/driver_fhkvdataiii.cc

And seems you are received too short telegram, please read about nanoCUL adapter, simila case.  You can try upgrade firmware, cannot help more.

mStirner, 2025-02-20:
How do i receive too short telegrams when the first one was received correctly?
I have read about the nanoCUL Adapter and that it has a too short buffer and need special firmware.

But how does that is related to my NOELEC adapter?
I contacted the Manufacturer and they said, "there is no firmware"

kzajac83, 2025-02-20:
Cannot explain and help more, also not know and not using your hardware, maybe @weetmuts can help/explain?

closed 2025-05-22 by github-actions[bot]