wmbusmeters/wmbusmeters #1492: Last Zeros are sometime behind sometime bevore comma

boyi01, 2025-02-17:
### Type of request

Bug

### HAOS version

14.2

### wmbusmeters version

1.18.0-26

### Your message goes here

[image]

We have a qheat connect via the addon.

Sometimes we got the Data like this : 805.1 kwh which is wrong, sometimes we get 8051 kwh which is right. 

The bug is in the mqtt, so no bug in how the numbers presented in ha, it's should be a bug how the data is written to mqtt

weetmuts, 2025-02-18:
Can you collect two telegrams that give the two different values?

schmyd, 2025-02-27:
Hi,

I think this isn't a comma issue, but related to different message types that yield a different result by a factor of 10. I propose to continue the discussion in #1497 where I've added a workaround and further analysis.

boyi01, 2025-02-28:
That sounds awesome, sadly i currently can't get the information due to holidays ...

closed 2025-02-28 by boyi01