wmbusmeters/wmbusmeters #1499: Support for the water meter Apator op-04-1a

belkop-ghb, 2025-03-03:
### Type of request

New device support

### Meter description

Hi, could you please add support for the water meter Apator op-04-1a?
Analysis showed that data are encrypted, but the driver is unknown.
I'm not sure what's the root cause of the problem.
Thank you a lot.

### Logged telegrams from meter in decrypted format

```shell
4e440106215224101a077a770040a5Ba7dd828e70af1256fee62e49dd24f85c7994de0ca8Bf7df03424284ad176Ba03c99766a52634edB849e374a318013a57e86197Bfc13599f6645f0d09f375ca5
```

bkralik, 2025-03-04:
It seems to me that your telegram is encrypted...

[image]

I also have Apator op-04-1a water meters (also called Naxom 1-a, [manufacturer page](https://www.apator.com/en/our-solutions/water-and-heat/remote-utility-reading-system/wireless-radio-system/op-o4-1-2-en#application)), but I was able to decrypt my telegrams. For example:

`3F440106967627101A077A6F0030A52F2F_0413D1010000046D23361B3C0F60000000000844058673F33B371E3BF70000004000000040000000400000002F2FEC`

And it seems that drive c5isf is able to decode some basic fields! [Look at analyzer](https://wmbusmeters.org/analyze/3F440106967627101A077A6F0030A52F2F0413D1010000046D23361B3C0F60000000000844058673F33B371E3BF70000004000000040000000400000002F2FEC)

Auto driver    : not found!
Similar driver : c5isf 08/43
Using driver   : c5isf 08/43
000   : 3f length (63 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : 0106 dll-mfct (APA)
004   : 96762710 dll-id (10277696)
008   : 1a dll-version
009   : 07 dll-type (Water meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : 6f tpl-acc-field
012   : 00 tpl-sts-field (OK)(OK)
013   : 30a5 tpl-cfg a530 (bidirectional synchronous AES_CBC_IV nb=3 cntn=0 ra=0 hc=0 )
015   : 2f2f already decrypted check bytes
017   : 04 dif (32 Bit Integer/Binary Instantaneous value)
018   : 13 vif (Volume l)
019 C!: D1010000 ("total_volume_m3":0.465)
023   : 04 dif (32 Bit Integer/Binary Instantaneous value)
024   : 6D vif (Date and time type)
025 C!: 23361B3C ("last_month_date":"2024-12-27 22:35")
029 C?: 0F manufacturer specific data 60000000000844058673F33B371E3BF70000004000000040000000400000002F2FEC

bkralik, 2025-03-04:
Btw, @belkop-ghb, it seems to me that my decryption key works for your telegram. I can share it, but please write me directly...

Encrypted:
`4e440106215224101a077a770040a5Ba7dd828e70af1256fee62e49dd24f85c7994de0ca8Bf7df03424284ad176Ba03c99766a52634edB849e374a318013a57e86197Bfc13599f6645f0d09f375ca5`

[Decrypted](https://wmbusmeters.org/analyze/4E440106215224101A077A770040A52F2F0413C3530000046D0F323A320F60000000000843327873F63B373F312B3C00006B20000024040000360000003600000036000000360000002F2F2F2F2F2F):
`4E440106215224101A077A770040A52F2F_0413C3530000046D0F323A320F60000000000843327873F63B373F312B3C00006B20000024040000360000003600000036000000360000002F2F2F2F2F2F`

belkop-ghb, 2025-03-11:
Great news @bkralik , I've sent you an email to get the decryption key. Thank you a lot, appreciate it.

kzajac83, 2025-03-11:
@bkralik I have Ultrimis from Apator and encrypted too telegrams, my local supply water company don't want share the key :(.  Can you share with me you key, maybe will works.  
I can also give my telegram if needed.

xmatekaj, 2025-03-12:
> Btw, [@belkop-ghb](https://github.com/belkop-ghb), it seems to me that my decryption key works for your telegram. I can share it, but please write me directly...

I wonder how this is possible as Apator says that each device has a unique key. 

I just got a new OP-04-1A device and received via mbus dongle two different frames. That puzzles me too... Will you be able to decrypt it as well?

12/03/202521:19:09.93:-33dBm APA FF47440106112250101A078CE0937AF37040A5CC2429B3821952141EA730AFC37CE4309674DB1613C318C9898E4C64CC5C5D9E7E496F7DC2C45208F3C9735938D7CDEF6DA6855ABCF8CD88FFA063B4F3A401C2B8

12/03/202521:31:41.04:-33dBm APA FF37440106112250101A078CE0947AF47030A59CFA5E02DD7276CC77733469B3693B5993D8EE65AB6BB77FCDADBD51DE0262FCE34057D796E46DF2AC7ACEF79B5ABAF9B8

bkralik, 2025-03-12:
> 
> I wonder how this is possible as Apator says that each device has a unique key.
> 

Each device has a key set by installation technician. It can be different for every installation (usually per building), but some local water meter installation company just decided to use ~16 keys that are hardcoded in their readout software. 😄 

But it seems to me that your telegrams are somehow deformed:

`$ wmbusmeters --analyze --verbose FF47440106112250101A078CE0937AF37040A5CC2429B3821952141EA730AFC37CE4309674DB1613C318C9898E4C64CC5C5D9E7E496F7DC2C45208F3C9735938D7CDEF6DA6855ABCF8CD88FFA063B4F3A401C2B8
(wmbusmeters) version: 1.18.0-27-g8aecb7b
(config) using device: <FF47440106112250101A078CE0937AF37040A5CC2429B3821952141EA730AFC37CE4309674DB1613C318C9898E4C64CC5C5D9E7E496F7DC2C45208F3C9735938D7CDEF6DA6855ABCF8CD88FFA063B4F3A401C2B8>:any
(config) number of meters: 0
Started config simulation listening on any
(simulation) in
(main) regular reset of simulation  will happen every 82800 seconds
(mbus) no 0x68 byte found, clearing buffer.
(serial) event loop stopped!
`
Do you have logtelegrams=true enabled? What dongle are you using?

bkralik, 2025-03-12:
@kzajac83 Post the telegram here or sent it to me directly and I will try to decode it 😄

xmatekaj, 2025-03-13:
> But it seems to me that your telegrams are somehow deformed:
>
> `$ wmbusmeters --analyze --verbose [...]

I'm using Adeunis Wmbus Usb Dongle. First FF should be omitted:
47440106112250101A078CE0937AF37040A5CC2429B3821952141EA730AFC37CE4309674DB1613C318C9898E4C64CC5C5D9E7E496F7DC2C45208F3C9735938D7CDEF6DA6855ABCF8CD88FFA063B4F3A401C2B8

kzajac83, 2025-03-13:
@bkralik some example from my Ultrimis below, maybe you could decode it. If need I have 64k telegrams stored one-be-one from about two months

```

2E4401065528088001167A44002005E12BA49A0082F4E3A687C8B28025231E3190E43737E05539CAFA111101C6F11A

2E4401065528088001167AA503200530271025832A7FAF849DBCDDE6C1E2CCD112D0CBA8DC2372D8515B3E2668C889

2E4401065528088001167A0C002005082AA5A3456D654EB56173CF3D8BB30A9E68D3031E235BA2CE2319187DEB328C

2E4401065528088001167A1C002005866CB876FC7A82221746B8658A1273DF66CB3EE015AF2F3F01BDA20D26453B8A
```

bkralik, 2025-03-14:
@xmatekaj I am sorry, but even then my key doesn't work - `failed decryption. Wrong key?`

@kzajac83 Unfortunately, same for you - `failed decryption. Wrong key?`

kzajac83, 2025-03-14:
thx @bkralik 

General I thinking there are some law in PL/EU that force water (or others) companies to share KEY on request?? Cannot understood blocking this.

smetaxa, 2025-03-16:
> [@xmatekaj](https://github.com/xmatekaj) I am sorry, but even then my key doesn't work - `failed decryption. Wrong key?`
> 
> [@kzajac83](https://github.com/kzajac83) Unfortunately, same for you - `failed decryption. Wrong key?`

Could you please try following data?

4e440106384921101a077a150040a51b3884834968fac5491fcc5195b94c69affe580ba081f754dee9c1e7a00834b715abc3b5236898a1750c852f0a03132551f4a9f47405ac09769abc4ded226720

Thank you in advance!

bkralik, 2025-03-16:
@smetaxa decoded: `4e440106384921101a077a150040a52f2f04131f2a0000046d193423330f60000000000805996973f73B373c3211290000331f00003B170000480f0000B1080000470000003a0000003a0000002f2f`
send me a direct mail...

mathijass, 2025-03-28:
Can anyone try decode? 

Received telegram from: 10531411
          manufacturer: (APA) Apator, Poland (0x601)
                  type: Water meter (0x07) encrypted
                   ver: 0x1a
                device: rtlwmbus[00000001]
                  rssi: 23 dBm
                driver: unknown!
telegram=|_41440106111453101A078CE0647A974030A5324A219975194724FA1C098EBAE065B5D944B82EA3EF619B1B0B1A3619765FA4C600193CF5BADB440522BE3A9BD5ADD9|+10

smetaxa, 2025-03-28:
@mathijass the key from @bkralik is not working for this data

nikolapleska, 2025-05-29:
Guys, can any of provide me with AES key in DM, please? Will try for Enbra (CZ) installation, they probably use shared key all over. Thanks.

pswiatki, 2025-06-10:
Decoding aside, this RSSI of 148 dBm is very much confusing. I understand where the number comes from, but the unit is wrong.

kzajac83, 2025-06-10:
> Can anyone try decode? [@bkralik](https://github.com/bkralik) [@smetaxa](https://github.com/smetaxa)
> 
> Received telegram from: 10701931 manufacturer: (APA) Apator, Poland (0x601) type: Water meter (0x07) encrypted ver: 0x1a device: rtlwmbus[00000001] rssi: 148 dBm driver: unknown! telegram=|_51440106311970101A078CC0157A8E7040854DF87B73EA1EF09CF034839ADC6238DA28B10428872E612337E7BE3E1532527B1CC8DA1E171F634C3A71599149499279BE71ED65CD8FFF82D57BAF1D399995E2|+62

What reader you use? I have module CC1101 with moxon antenna, max up to 90 dBm.

pswiatki, 2025-06-10:
90 dBm is also insane. Doesn't make any sense. (it's the unit, not the value).

kzajac83, 2025-06-10:
Up to **minus** 90 dBm in my case.

pswiatki, 2025-06-10:
That (sign) would make all the difference. Although, consequently, -148 dBm couldn't possibly be reliably "measured" by a regular receiver, such as RTL-SDR, FIR example. It would be too weak.

djmalin, 2025-06-11:
> What reader you use? I have module CC1101 with moxon antenna, max up to 90 dBm.

RTL2838UHIDIR

belkone, 2025-08-15:
Hi, can anyone try to decode? :)

Received telegram from: 10891172
         manufacturer: (APA) Apator, Poland (0x601)
                 type: Water meter (0x07) encrypted
                  ver: 0x1a
               device: rtlwmbus[00000001]
                 rssi: 90 dBm
               driver: unknown!
telegram=|_41440106721189101A078CE0287AD63030A5CC77537D9D9E36D35E3DD8BE307A7356B5F41DAEEC6F96CD536A902266F25B7947F7C02986E5093E998D935A6C5ECB40|+46

Received telegram from: 11047085
         manufacturer: (APA) Apator, Poland (0x601)
                 type: Water meter (0x07) encrypted
                  ver: 0x1a
               device: rtlwmbus[00000001]
                 rssi: 73 dBm
               driver: unknown!
telegram=|_41440106857004111A078CE0847AA13030A5294DD73956915AB6F10D1848C461E6C68AA13DA2F9DD7D7608E72F0E61881543866B4F96C8634456CCB110D3642755FF|+216

city12001, 2025-08-21:
Hi, can anyone try to decode? :)

Received telegram from: 80033850
          manufacturer: (APA) Apator, Poland (0x601)
                  type: Cold water meter (0x16) encrypted
                   ver: 0x20
                device: iu891a[00202083]
                  rssi: -73 dBm
                driver: unknown!
telegram=|_434401065038038020168C0055900F002C259CC5A6684501188CE963F2E87ACB00200710D39A2A65AF956488381368E26263CDBE65717BC965F91F0EE6C36A0BAA18131E|+14
Received telegram from: 80033850
          manufacturer: (APA) Apator, Poland (0x601)
                  type: Cold water meter (0x16) encrypted
                   ver: 0x20
                device: iu891a[00202083]
                  rssi: -73 dBm
                driver: unknown!
telegram=|_434401065038038020168C0055900F002C25ACC5A66827E4D592AE368A797ACC00200710EFD5CA421132156D7C587032CA3BD4B7CA34E423A662A14280342EA8C07F7B69|+31
Received telegram from: 51426439
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Smoke detector (0xf0)
                   ver: 0x76
                device: iu891a[00202083]
                  rssi: -100 dBm
                driver: tsd2
telegram=|_294468503964425176F0A0009F31123300C0123300808A0102051A17631B1763008DC3009ED4000FE500|+35
Received telegram from: 80033850
          manufacturer: (APA) Apator, Poland (0x601)
                  type: Cold water meter (0x16) encrypted
                   ver: 0x20
                device: iu891a[00202083]
                  rssi: -73 dBm
                driver: unknown!
telegram=|_434401065038038020168C0055900F002C25BCC5A668143E4AF421A959C87ACD00200710FFA0FBF4CBEA3C1EC75C9C7B46EAEAE4F097781455F1DEF8F07792C7C68A16EF|+46

Thanks a lot for your help in advance

best regards

Klaus

kuchar91, 2025-08-29:
Can someone try to decode my frames? 

`Meter ID [0x10527532] RSSI: -62 dBm LQI: 136 Frame: T1 A not found in configuration T: 61440106327552101A078CC0BF7AF9705085CAD957EA88E0F8D805B13ECF778C5C9FBFD033B0B249EB70495193D096F93FED5398CC2997C570CD9F6E86D4633D36488191C80277AF710F22C434159086DF66AD3BBDB78E8971529FF6A66EA19ED518 (98)`

`Meter ID [0x10740600] RSSI: -58 dBm LQI: 129 Frame: T1 A not found in configuration T: 41440106000674101A078CC0A17AE6003085C91959FE3C75D75FE707467855365B58FDA8E2F7A18C97F1E1E544A60E006A8D806FB9AC132509CFB4D85EF98530F80A (66)`

mmmalina, 2025-11-03:
I also have an OP-04-1a, and like @kitor , my meter sends both encrypted and unencrypted data. Unfortunately, I can't get the meter reading for unencrypted data. Will I need to obtain meter keys from the meter manager even if the meter sends unencrypted data?

elgatho, 2025-11-27:
@bkralik  @smetaxa just got new meter, kindly please, try to check
61440106065977101A078CC0207A807050853C06CF442DC0BDBBBEDCBF456125C0527AFF913287D9BA4280588DD7E85CB82848A61243B4631974D0980A95FB961EF983519E3FC472BC74B9A01318CD278F9FB635B29FE99D6F84B703D8B783C4F8F3

51440106065977101A078CC0257AB57040855F102059640671E8DBA49DAA9684D9EFB8A70E8A9CF31801FC0F10453C2113B5CED579FC01CC0140189A2F54E49DC2798066E2EC8FE7E9AC154C410623E80C88

kzajac83, 2026-02-06:
@bkralik can you share the software or provide link where I can download it?

closed 2026-09-20 by github-actions[bot]