wmbusmeters/wmbusmeters #1500: Receive but not decode Heater Telegrams

GAS85, 2025-03-05:
### Type of request

Question

### OS version

Ubuntu 24.04

### wmbusmeters version

1.83

### Your message goes here

Hello dear community,

I have few devices on a place with my personal AES keys, I can decode payload very well. Somehow I can't see any data from my heater. Following Telegram is send:

```plane
ad44c5147750854100047a120090254aa279de4f63df96a7B0516439feed192219533e2B0Bf1c187076246c6e3f461f90c8687520B9c18aaa1c497e5B54cdde2e8e4298f0889f3aa4de43aB1150f5151c5aca75adedda6aB1f8783614e2a08cdBa1a6dcB9B6f3a5911ef5f14a3f4B3a6c83fe6c919a3c33f7e15c538381612B43a1f7c2adcf5334f1d8737d5ee2f15ac781522c3d128719B13e71560de4463e60d03fd0c05010002fd0B111102ea
```

I use following config for it:

```bash
name=Heizung
id=41855077
key=<SOMEKEY>
driver=sensostar
```

it is got decoded under https://wmbusmeters.org/analyze/ as

```
Auto driver    : sensostar
Similar driver : microclima 55/81
Using driver   : sensostar 00/00
000   : ad length (173 bytes)
001   : 44 dll-c (from meter SND_NR)
002   : c514 dll-mfct (EFE)
004   : 77508541 dll-id (41855077)
008   : 00 dll-version
009   : 04 dll-type (Heat meter)
010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
011   : 12 tpl-acc-field
012   : 00 tpl-sts-field (OK)(OK)
013   : 9025 tpl-cfg 2590 (synchronous AES_CBC_IV nb=9 cntn=0 ra=0 hc=0 )
015   : 2f2f decrypt check bytes (OK)
017   : 04 dif (32 Bit Integer/Binary Instantaneous value)
018   : 6D vif (Date and time type)
019 C!: 3A302533 ("meter_datetime":"2025-03-05 16:58")
023   : 04 dif (32 Bit Integer/Binary Instantaneous value)
024   : 06 vif (Energy kWh)
025 C!: 7E0F0000 ("total_kwh":3966)
029   : 04 dif (32 Bit Integer/Binary Instantaneous value)
030   : 13 vif (Volume l)
031 C!: 1A2A0300 ("total_water_m3":207.386)
035   : 01 dif (8 Bit Integer/Binary Instantaneous value)
036   : FD vif (Second extension FD of VIF-codes)
037   : 17 vife (Error flags (binary))
038 C!: 00 ("status":"OK")
039   : 42 dif (16 Bit Integer/Binary Instantaneous value storagenr=1)
040   : 6C vif (Date type G)
041 C!: 1F3C ("target_date":"2024-12-31")
043   : 44 dif (32 Bit Integer/Binary Instantaneous value storagenr=1)
044   : 06 vif (Energy kWh)
045 C!: A0040000 ("target_kwh":1184)
049   : 84 dif (32 Bit Integer/Binary Instantaneous value)
050   : 01 dife (subunit=0 tariff=0 storagenr=2)
051   : 06 vif (Energy kWh)
052 C!: E00E0000 ("target_2_kwh":3808)
056   : 84 dif (32 Bit Integer/Binary Instantaneous value)
057   : 02 dife (subunit=0 tariff=0 storagenr=4)
058   : 06 vif (Energy kWh)
059 C!: 600A0000 ("target_4_kwh":2656)
063   : 84 dif (32 Bit Integer/Binary Instantaneous value)
064   : 03 dife (subunit=0 tariff=0 storagenr=6)
065   : 06 vif (Energy kWh)
066 C!: A0040000 ("target_6_kwh":1184)
070   : 84 dif (32 Bit Integer/Binary Instantaneous value)
071   : 04 dife (subunit=0 tariff=0 storagenr=8)
072   : 06 vif (Energy kWh)
073 C!: 00000000 ("target_8_kwh":0)
077   : 84 dif (32 Bit Integer/Binary Instantaneous value)
078   : 05 dife (subunit=0 tariff=0 storagenr=10)
079   : 06 vif (Energy kWh)
080 C!: 00000000 ("target_10_kwh":0)
084   : 84 dif (32 Bit Integer/Binary Instantaneous value)
085   : 06 dife (subunit=0 tariff=0 storagenr=12)
086   : 06 vif (Energy kWh)
087 C!: 00000000 ("target_12_kwh":0)
091   : 84 dif (32 Bit Integer/Binary Instantaneous value)
092   : 07 dife (subunit=0 tariff=0 storagenr=14)
093   : 06 vif (Energy kWh)
094 C!: 00000000 ("target_14_kwh":0)
098   : 84 dif (32 Bit Integer/Binary Instantaneous value)
099   : 08 dife (subunit=0 tariff=0 storagenr=16)
100   : 06 vif (Energy kWh)
101 C!: 00000000 ("target_16_kwh":0)
105   : 84 dif (32 Bit Integer/Binary Instantaneous value)
106   : 09 dife (subunit=0 tariff=0 storagenr=18)
107   : 06 vif (Energy kWh)
108 C!: 00000002 ("target_18_kwh":33554432)
112   : F0 dif (No data Value during error state storagenr=1)
113   : 53 dife (subunit=1 tariff=1 storagenr=7)
114   : D0 vif (Mass g/h)
115   : DF combinable vif (DurationExceedsUpperLast)
116   : 6E combinable vif (DateTimeOfDELastBegin)
117   : E2 dif (16 Bit Integer/Binary Minimum value storagenr=1)
118   : BA dife (subunit=0 tariff=3 storagenr=21)
119   : EA dife (subunit=2 tariff=11 storagenr=341)
120   : C4 dife (subunit=6 tariff=11 storagenr=2389)
121   : 69 dife (subunit=14 tariff=139 storagenr=76117)
122   : 6D vif (Date and time type)
123 C?: 6B85
125   : E4 dif (32 Bit Integer/Binary Minimum value storagenr=1)
126   : D5 dife (subunit=1 tariff=1 storagenr=11)
127   : DC dife (subunit=3 tariff=5 storagenr=395)
128   : 54 dife (subunit=7 tariff=21 storagenr=2443)
129   : 07 vif (Energy 10⁴ Wh)
130 C?: B762909C
134   : 4B dif (6 digit BCD Instantaneous value storagenr=1)
135   : 69 vif (Pressure 10⁻² bar)
136 C?: FAA0AA
139   : 56 dif (48 Bit Integer/Binary Maximum value storagenr=1)
140   : 45 vif (Volume flow ext. 10⁻² m³/min)
141 C?: 70CB2DA9DAA3
147   : 7A dif (4 digit BCD Value during error state storagenr=1)
148   : 5F vif (Return temperature °C)
149 C?: 366F
151   : 61 dif (8 Bit Integer/Binary Minimum value storagenr=1)
152   : 54 vif (Mass 10¹ kg/h)
153 C?: F0
154   : 4B dif (6 digit BCD Instantaneous value storagenr=1)
155   : EA vif (Pressure 10⁻¹ bar)
156   : BE combinable vif (ValueAtBaseCondC)
157   : 8B combinable vif (Reserved)
158   : BF combinable vif (ObisDeclaration)
159   : E6 combinable vif (DurationOfDLast)
160   : 0D combinable vif (Reserved)
161 C?: 03FD0C
164   : 05 dif (32 Bit Real Instantaneous value)
165   : 01 vif (Energy 10⁻² Wh)
166 C?: 0002FD0B
170   : 11 dif (8 Bit Integer/Binary Maximum value)
171   : 11 vif (Volume 10⁻⁵ m³)
172 C?: 02
173   : EA dif (4 digit BCD Minimum value storagenr=1)

{
    "_":"telegram",
    "media":"heat",
    "meter":"sensostar",
    "name":"",
    "id":"41855077",
    "target_kwh":1184,
    "target_date":"2024-12-31",
    "target_10_kwh":0,
    "target_12_kwh":0,
    "target_14_kwh":0,
    "target_16_kwh":0,
    "target_18_kwh":33554432,
    "target_2_kwh":3808,
    "target_4_kwh":2656,
    "target_6_kwh":1184,
    "target_8_kwh":0,
    "total_kwh":3966,
    "total_water_m3":207.386,
    "meter_datetime":"2025-03-05 16:58",
    "status":"OK",
    "timestamp":"2025-03-05T16:00:03Z"
}

Using: wmbusmeters: 1.17.1-70-g41a2b48
41a2b4808f00367af7bbaac57e221539825babbb
```

But in a logs I can see only this:

```
(cul) checkCULFrame LQI=64
(wmbus) trimming frame B "AD44C5147750854100047A120090254AA279DE4F63DF96A7B0516439FEED192219533E2B0BF1C187076246C6E3F461F90C8687520B9C18AAA1C497E5B54CDDE2E8E4298F0889F3AA4DE43AB1150F5151C5ACA75ADEDDA6AB1F8783614E2A08CDBA1A6DCB9B6F3A5911EF5F14A3F4B3A6C83FE6C919A3C33F7E15C538381612B43A1F7C2ADCF5334F1D8737D5EE2F15AC781522C3D128719B13E71560DE4463E60D03FD0C05010002FD0B111102EA"
(wmbus) ff b dll crc first 0-126 12b4 ok
(wmbus) ff b dll crc (calculated 3569) did not match (expected 02ea) for bytes 128-172!
(cul) dll C1 (frame b) crcs failed check! Ignoring telegram!
(cul) error in received message.
```

IDK what to do with it...

GAS85, 2025-03-06:
Not sure, but probably it is related to #953

GAS85, 2025-05-06:
Not stale

GAS85, 2025-05-06:
👍

dartrax, 2026-01-11:
> Not sure, but probably it is related to [#953](https://github.com/wmbusmeters/wmbusmeters/issues/953)

I also do think it is the same problem. Let's see if there are any news over there.