wmbusmeters/wmbusmeters #1501: brummerhoop (engelmann) waterstar m cold water meter does not send any telegram

apf-doit, 2025-03-06:
### Type of request

Question

### HAOS version

14.2

### wmbusmeters version

1.18.0-RC1

### Your message goes here

Hi there, i am running an iperl sensus on a nano CUL 868 MBUS with this add-on and can read all the values while decrypting works just fine. But recently i bought another meter for outside - a brummerhoop waterstar m which is just a rebranded engelmann.

Since then i am very frustrated for over a few days now since i installed the meter physically while having no success to see any telegram of that device. wireless radio (868MHz) is enabled on the device, I tried 3 different devices of the same type including 2 different versions of firmware, but all no luck. i suspect all the devices are simply not sending anything.

In case I am right I would like to ask the advanced users here on their opinion if interpreting the manual is correct:

[image]

Does that mean if all is factory default:

1.  default long telegramm -> walk-by: is a different type of telegram which i can't see even if I'm listening on the correct frequency?
2. default master key: i have a device key in the package - is it not using it instead it is sending encrypted telegrams with the master key which I don't know of course? But as I said before I don't see anything, not even an encrypted telegram.
3.  I obvisiouly can not change the radio settings on the device and need an optical reader and the engelmann software for that which would increase the costs stupidly? Anyone here knows about if this is the case?

I would highly appreciate any comment just to know which direction that goes, if all is true what i suspect this device is useless for an end customer.

best regards
Alex

apf-doit, 2025-03-06:
this is my config in /homeassistant/wmbusmeters/etc/wmbusmeters.d/

meter-0001:

```
driver=iperl
id=22600228
key=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
name=sensus
```

meter-0002:

```
driver=waterstarm
id=23260498
key=xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
name=brummerhoop
```

since the last digits of the serial number of the iperl is used to be the id, i just used the serial number of the brummerhoop as the is as well. Unfortunately i can not sniff on the device because i don't know how since this is the add-on and not the binary itself. If there is a way please feel free to tell me.

apf-doit, 2025-03-07:
I connected the nano CUL 868 MBUS USB stick to another pi and followed the instruction to make the received telegrams visible:

```
root@pi3:~ # wmbusmeters auto:t1

Started auto cul on /dev/ttyUSB0 listening on t1
No meters configured. Printing id:s of all telegrams heard!

Received telegram from: 22600228
          manufacturer: (SEN) Sensus Metering Systems, Germany (0x4cae)
                  type: Water meter (0x07) encrypted
                   ver: 0x68
                device: cul
                  rssi: -54 dBm
                driver: iperl
```

So i guess I am right that the brummerhoop waterstar m devices not sending anything.

Anyone here having such a device and can advise how to make it send telegrams?

apf-doit, 2025-03-07:
In case the other devices sending not T1, i start to listen with that command:

`root@pi3:~ # wmbusmeters --debug auto:t1,c1`

but then i get:

```
(main) opening t1,c1
Started auto cul on /dev/ttyUSB0 listening on t1,c1
(cul) on /dev/ttyUSB0
(serial) EVENT thread interrupted
(serial) before config: /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serial) after config:  /dev/ttyUSB0 speed(B38400) input() output() control(CLOCAL|CREAD|CSIZE) local() special_chars() modem(DTR|RTS)
(serialtty) opened /dev/ttyUSB0 fd 8 (cul)
(main) regular reset of cul  on /dev/ttyUSB0 will happen every 82800 seconds
(cul) getDeviceId
Warning! Desired link modes t1,c1 cannot be set for device /dev/ttyUSB0:cul[]
```

Ich checked all issues here and came across the firmware, but I have the latest already:

`(cul) probe response "V 1.67 nanoCUL868_r568<0D><0A>"`

running out of ideas...

apf-doit, 2025-03-07:
Ah, That is something new I have not seen before:

```
(cul) Warning! Your cul firmware has a bug that prevents longer telegrams from being received.!
(cul) Please read: https://github.com/wmbusmeters/wmbusmeters/issues/390
(cul)         and: https://wmbusmeters.github.io/wmbusmeters-wiki/nanoCUL.html
(cul) bad hex "A944FA12980426230207C9997AE40090254A6711370C5D9BA82DC43AD5F0DB435A2496DF4AE4B75F5CA2FDA0CD3B76A50D129EB378D83F1614F8DE19061BBA790E76203C17A2D4F19E708DE6B3E05930132B213EE4115834D51D6E09AB80D5726C51741B078A15A6907EC6E3990E34CF5A3D3D55BE4C6E1EA7F584E9AAC12E6DF894A77686B7E323B32E5BEFC4DBE799B8224"
(cul) warning: the hex string is not proper! Ignoring telegram!
(cul) error in received message.
```
could this be my missing telegrams?

hn, 2025-03-10:
By factory default, the M-ETH watermeter only sends data during the day from 08:00-18:00.

For me, it nicely works with ESPHome and an CC1101 receiver, see [here](https://github.com/hn/esphome-configs/blob/master/watermeter-waterstarm/watermeter-waterstarm.yaml) and [here](https://github.com/SzczepanLeon/esphome-components/issues/192).

apf-doit, 2025-03-12:
Hi @hn , thx for your response. I was thinking about ESPHome instead of wmbusmeter, but for that i would need a ESP32 - which i don't have. And I don't want to buy another receiver. It becomes way to costy for just for fun.

I know about all the features like sending timeslots etc. of the engelmann, I just don't see any telegram. If my setup would be faulty I couldn't receive the telegrams from the iPerl meter. I followed exactly the docs how to enable radio on the device and I see the bar in the display. Did I miss anything - how did you turned on radio?

hn, 2025-03-14:
The water meter was radio enabled when it was delivered by AMZN (at least I didn't knowingly turn it on).

The only challenge was that the first time I tried it was in the evening and received no signals. And in the morning it magically worked :)

Marsen76, 2025-03-17:
Hello @apf-doit, the messages from the Waterstar are too long for the nanoCUL. The buffer from the cul is not big enough. I had the same issue in ioBroker.

Sorry for my bad english.

apf-doit, 2025-03-17:
Oh. Thats not a good message. That would require a new receiver setup. Man, I was so glad to get the iperl up and running so easy with the nanCUL but didn't expect it be so difficult to set up another meter. 

Could you recommend a ready made product which can deal with both of the meters?

Marsen76, 2025-03-17:
I'm trying to create a ESP32 with ESPhome.
I will report later.

> For me, it nicely works with ESPHome and an CC1101 receiver, see [here](https://github.com/hn/esphome-configs/blob/master/watermeter-waterstarm/watermeter-waterstarm.yaml) and [here](https://github.com/SzczepanLeon/esphome-components/issues/192).

apf-doit, 2025-03-28:
Just for the records: I am so frustrated that I've purchased a IMST iU891A-XL USB stick. That device comes with a nice software for configuration and a ready to use packet scanner which scans all modes simultaneously. I can see all kinds of telegrams: from my iPerl and all the funny Techem devices all around the neighborhood. But clearly not the Brummerhoop. That device simply does not send shit even though radio seems to be on. Regarding the MAN code table it is pretty easy to identifiy any vendor sending something:

https://www.m-bus.de/man.html

I have contacted the vendor for an explaination, obvisiouly it requires the optohead module for a decent config since the display does not support any important setup.

apf-doit, 2025-04-01:
Another one for the records: vendor replied and told me the real manufacturer id - it is not an Engelmann device, it is a Lorenz device (alpha: DWZ - hexa: 12FA). When you know the MAN code, you can calculate the HEX with that formula:

`MAN = [ASCII(1. letter)-64] * 32 * 32 + [ASCII(2. letter)-64] * 32 + [ASCII(3. letter)-64]`

more MAN codes are available here: 

https://github.com/CBrunsch/WMBus-Sniffer-MUC/blob/master/mbus/manufacturers.txt

That being said, i now see the water meter with the IMST iU891A-XL USB stick - but never saw it with the nano CUL 868 MBUS USB stick. I never paid attention to that ID since i was looking for the wrong one. The radio ist setup almost like the manual says - except  in wrong mode: C1 instead of T1. The vendor also offered to rent to optohead device for free to make the setup final for my needs.

I probably will stick with the IMST iU891A-XL and report later if the add-on can catch all the telegrams i want.

What a journey...

apf-doit, 2025-04-06:
Last Update: I got the optohead for the Lorenz meter and configured radio to my taste. All working now, I can read both water meters on CLI or HA. I will sell the nanoCUL and stick with the iU891A-XL since that device did not give me headaches at all.

closed 2025-04-06 by apf-doit

hn, 2025-04-07:
@apf-doit Thanks for the explanations. I would also like to change the settings of my Brummerhoop water meter. If you have an optohead, how do you send the commands to do this? There is [commercial software](https://www.engelmann.de/product/device-monitor/) available [to buy](https://zaehleronlineshop.de/?a=324&lang=eng), but it's not really worth it for me (4x the cost of the water meter). And there also is some kind of [web configurator](https://konfigurator.lorenz-meters.de/).

Have you contacted Brummerhoop, Engelmann or Lorenz support?

apf-doit, 2025-04-07:
@hn I've contacted Brummerhoop directly. Usually you can rent the optohead for a week from their website including a link to the software to read and write to the meter using the serial COM port. They sent me the optohead for free with a return label, after explaining my situatuon. Just shipping you pay once. It doesnt have to be the same in your case. Just ask nicely ;)

Fun fact: after setting the telegram mode to short even the nanoCUL can read them now. So all the problems started with using a receiver setup with the known limitations.

apf-doit, 2025-04-07:
I changed the title of the issue to make it better findable for others.

hn, 2025-04-08:
Friendly Brummerhoop support send my this [offer](https://www.brummerhoop.com/miete-usb-optokopf-fuer-energiezaehler/50041745-M) which is twice the cost of the water meter. And I don't actually need the optohead hardware. Have to think about that ...

apf-doit, 2025-04-10:
i've contacted the seller via amzn. maybe you try this again. being monitored by the selling platform can make a difference.

juergen908, 2025-04-11:
> Last Update: I got the optohead for the Lorenz meter and configured radio to my taste. All working now, I can read both water meters on CLI or in HA. I will sell the nanoCUL and stick with the iU891A-XL since that device did not give me headaches at all.

Könntest du mich bzgl. des Brummerhoop kontaktieren ? Danke !

apf-doit, 2025-04-15:
@juergen908 what would you like to know?

hn, 2025-04-16:
@apf-doit What's the name of the software they send you with the optohead? I can't find the link on their website.

apf-doit, 2025-04-20:
The software comes with the optohead as QR code. I am not sure if the software is public or just comes with the rental. I can give the url in a private message. But i don't see any private here.

hn, 2025-04-20:
@apf-doit Github has no PM. Please send me an email, my email is my firstname at my lastname dot de. Thanks.

apf-doit, 2025-04-20:
@Marsen76 i can send you the download link by email if you like.

Marsen76, 2025-04-20:
That would be nice.
My adress is 
m_ss
at
gmx
.de

juergen908, 2025-04-22:
> [@Marsen76](https://github.com/Marsen76) i can send you the download link by email if you like.

Me too please.
j.bitzer
at
t-online
.de
Thanx !

hn, 2025-05-18:
One can use [this tool](https://github.com/hn/em-admin) to read and change the watermeter's radio parameters.

NigiNagiNu, 2025-07-01:
> [@Marsen76](https://github.com/Marsen76) i can send you the download link by email if you like.

I'm also playing around with a Lorenz water meter. The mentioned github tool didn't work out for me. So I'd like to give the official tool a try. Could you please:
michael.froestl
at
imail
.de
Many thanks upfront!

t509, 2025-08-04:
> [@Marsen76](https://github.com/Marsen76) i can send you the download link by email if you like.

@apf-doit would be very nice if you share the with mee too p955i (at) proton.me

ChristophWeigert, 2025-09-25:
Hi, I got the same issue as op here. Would it work to reprogram the meter with something like this? https://www.amazon.de/Hichi/dp/B0CH95T7X4?th=1 I have this in use currently to track my power meter. 

If someone could also send me the link to the requried software that would be nice. mail is like my github handle chr...ph@we...rt.cc (cc intended instead of de)