wmbusmeters/wmbusmeters #1503: ESPHome wmbus component fails after ESPHome 2024.11.3 update

FARKIr, 2025-03-21:
### Type of request

Enhancement

### HAOS version

2024.11.3

### wmbusmeters version

oldrer

### Your message goes here

Description:
I am experiencing an issue with the wmbus component after updating to ESPHome 2024.11.3. The component was working correctly before, but now I'm getting the following error:

I didn't change anything in my configuration. It appears that in ESPHome 2024.11.3, the 'ipv4address' validation method might have been replaced or renamed, causing the wmbus component to fail.

Could you please update the component to be compatible with the latest ESPHome version?

System information:
- ESPHome version: 2024.11.3
- Component path: /data/external_components/5c086c68/components/wmbus/

`INFO ESPHome 2024.11.3
INFO Reading configuration /config/esphome/vodomer.yaml...
ERROR Unable to load component wmbus:
Traceback (most recent call last):
  File "/esphome/esphome/loader.py", line 177, in _lookup_module
    module = importlib.import_module(f"esphome.components.{domain}")
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/data/external_components/5c086c68/components/wmbus/__init__.py", line 75, in <module>
    cv.Required(CONF_IP_ADDRESS):                 cv.ipv4address,
                                                  ^^^^^^^^^^^^^^
AttributeError: module 'esphome.config_validation' has no attribute 'ipv4address'
Failed config

wmbus: [source /config/esphome/vodomer.yaml:39]
  
  Component not found: wmbus.
  clients: 
    - name: wmbusmeters
      ip_address: 192.168.1.120
      port: 7011
  mosi_pin: GPIO13
  miso_pin: GPIO12
  clk_pin: GPIO14
  cs_pin: GPIO2
  gdo0_pin: GPIO5
  gdo2_pin: GPIO4`

Yoobe, 2025-04-07:
same here with 2024.10.2

SzczepanLeon, 2025-04-09:
Not related to wmbusmeters. Please update ESPHome.

closed 2025-05-06 by FARKIr