wmbusmeters/wmbusmeters #1509: Add MQTT autodiscovery for tariff-based energy sensors in Apator EMU driver (amiplus).

Mice30, 2025-04-03:
Currently, the amiplus driver successfully decodes individual tariff values (total_energy_consumption_tariff_{1,2,3}_kwh and total_energy_production_tariff_{1,2,3}_kwh), but MQTT autodiscovery definitions for these fields are missing.

Each update to wmbusmeters requires manually editing the autodiscovery JSON configuration to restore these sensors in Home Assistant.

Proposal: Add MQTT autodiscovery definitions for each tariff as separate energy sensors:

    total_energy_consumption_tariff_1_kwh

    total_energy_consumption_tariff_2_kwh

    total_energy_consumption_tariff_3_kwh

    total_energy_production_tariff_1_kwh

    total_energy_production_tariff_2_kwh

    total_energy_production_tariff_3_kwh

Example JSON configuration:
`"total_energy_consumption_tariff_1_kwh": {
    "component": "sensor",
    "discovery_payload": {
        "name": "Total energy consumption tariff 1",
        "unique_id": "wmbusmeters_{id}_total_energy_consumption_tariff_1_kwh",
        "state_topic": "wmbusmeters/{name}",
        "value_template": "{{ value_json.total_energy_consumption_tariff_1_kwh }}",
        "unit_of_measurement": "kWh",
        "icon": "mdi:transmission-tower-import",
        "device_class": "energy",
        "state_class": "total_increasing",
        "enabled_by_default": true,
        "device": { ... }
    }
},
`

Analogous entries should be added for tariffs 2 and 3 and for production sensors.

Benefits:

    Persistent sensors across updates.

    Native support for multi-tariff meters.

Additional context: Attached a decoded telegram example
[telegram.txt](https://github.com/user-attachments/files/19592703/telegram.txt)

[amiplus_json_enchenced.txt](https://github.com/user-attachments/files/19592850/amiplus_json_enchenced.txt)

https://wmbusmeters.org/analyze/8e4401062125665601027a671080052f2f066d5e5f142334800c78212566560afdc9fc0140020afdc9fc0237020afdc9fc0332028e30833c0000000000008e20833c0000000000008e10833c3191171000000BaBc8fc100001008e10032775650500008e20030000000000008e30030000000000000B2B8003000BaB3c0000002f2f2f2f2f2f2f2f2f2f2f2f2f2f2f

darmach, 2025-04-04:
@Mice30 I linked this issue to the PR, but realized that I'm not sure if that's what you mean. wmbusmeters-ha-addon is the correct place to fix that, right? Or you meant to fix it different way? (is there a different way?)

Mice30, 2025-04-04:
@darmach Yes, I’ve created a bit of a mess here—need to clean it up 😅
In short: when I scan my telegram at wmbusmeters.org, two properties are not recognized by the driver. This isn’t related to the HA add-on, just the core wmbusmeters driver itself.

closed 2025-04-22 by BIBOLV