wmbusmeters/wmbusmeters #1517: short telegram from Engelmann Waterstar M using OMS Modus C1

UBessle, 2025-04-23:
I have Waterstar M water meters from Engelmann, that send our short telegrams using OMS-Modus C1 but can't be decrypted by wmbusmeters with waterstarm driver

https://wmbusmeters.org/analyze/4144c5140124065070068c20e97a260030258aa4fa4eeaBB7ff16a3d4d4d195f6905291547f08e0dcad6d16B6730f3366256add4362B3642294B7f6d2B703ec1291e8ca7f4:auto:32E0B97E644EBB561ED485F36F2B3B7A 

I get the following debug log output during processing a telegram from this water meter
(serial) received binary "A5C2034144C5140124065070068C20E97A260030258AA4FA4EEABB7FF16A3D4D4D195F6905291547F08E0DCAD6D16B6730F3366256ADD4362B3642294B7F6D2B703EC1291E8CA7F4"
(im871a) checkIM871AFrame "A5C2034144C5140124065070068C20E97A260030258AA4FA4EEABB7FF16A3D4D4D195F6905291547F08E0DCAD6D16B6730F3366256ADD4362B3642294B7F6D2B703EC1291E8CA7F4"
(im871a) has_timestamp=0 has_rssi=1 has_crc16=1
(im871a) endpoint 2
(im871a) msgid 3
(im871a) rssi 140 (-52 dBm)
(im871a) got crc16 f4a7 expected f4a7
(im871a) received full frame
(wmbus) parseDLL @0 66
(wmbus) parseELL @10 56
(wmbus) parseNWL @13 53
(wmbus) parseAFL @13 53
(wmbus) parseTPL @13 53
(meter) Heizkreis-heat: for me? 50062401.M=EFE.V=70.T=06 in 83017974
(meter) Heizkreis-heat: not for me: no match
(wmbus) parseDLL @0 66
(wmbus) parseELL @10 56
(wmbus) parseNWL @13 53
(wmbus) parseAFL @13 53
(wmbus) parseTPL @13 53
(meter) Hygienespeicher-heat: for me? 50062401.M=EFE.V=70.T=06 in 82987783
(meter) Hygienespeicher-heat: not for me: no match
(wmbus) parseDLL @0 66
(wmbus) parseELL @10 56
(wmbus) parseNWL @13 53
(wmbus) parseAFL @13 53
(wmbus) parseTPL @13 53
(meter) Pufferspeicher-heat: for me? 50062401.M=EFE.V=70.T=06 in 82987784
(meter) Pufferspeicher-heat: not for me: no match
(meter) no meter handled 50062401.M=EFE.V=70.T=06 checking 8 templates.
(wmbus) parseDLL @0 66
(wmbus) parseELL @10 56
(wmbus) parseNWL @13 53
(wmbus) parseAFL @13 53
(wmbus) parseTPL @13 53
(meter) Heizkreis-heat: for me? 50062401.M=EFE.V=70.T=06 in 83017974
(meter) Heizkreis-heat: not for me: no match
(meter) Pufferspeicher-heat: for me? 50062401.M=EFE.V=70.T=06 in 82987784
(meter) Pufferspeicher-heat: not for me: no match
(meter) Warmwasser-VL-Volumen: for me? 50062401.M=EFE.V=70.T=06 in 23043440
(meter) Warmwasser-VL-Volumen: not for me: no match
(meter) WP-Puffer-VL: for me? 50062401.M=EFE.V=70.T=06 in 50062401
(meter) WP-Puffer-VL: meter detection did not match the selected driver waterstarm! correct driver is: unknown!
(meter) Not printing this warning again for id: 50062401 mfct: (EFE) Engelmann Sensor, Germany (0x14c5) type: Warm Water (30°C-90°C) meter (0x06) ver: 0x70
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) WP-Puffer-VL: yes for me
(formula) parsing "storage_counter - 1 counter"
(formula) tokens: FIELD(storage_counter) MINUS(-) NUMBER(1) UNIT(counter) 
(formula) handle dventry field storage_counter into storage counter
(formula) <SUB <DVENTRY storage_counter> <CONST 1 counter[1]Dimensionless> >
(formula) parsing "meter_datetime - ((storage_counter-1counter) * 1 month)"
(formula) tokens: FIELD(meter_datetime) MINUS(-) LPAR(() LPAR(() FIELD(storage_counter) MINUS(-) NUMBER(1) UNIT(counter) RPAR()) TIMES(*) NUMBER(1) UNIT(month) RPAR()) 
(formula) handle meter field meter_datetime into meter datetime
(formula) handle dventry field storage_counter into storage counter
(formula) <SUB <FIELD meter_datetime> <TIMES <SUB <DVENTRY storage_counter> <CONST 1 counter[1]Dimensionless> > <CONST 1 month[1month]Time> > >
(formula) parsing "storage_counter - 1 counter"
(formula) tokens: FIELD(storage_counter) MINUS(-) NUMBER(1) UNIT(counter) 
(formula) handle dventry field storage_counter into storage counter
(formula) <SUB <DVENTRY storage_counter> <CONST 1 counter[1]Dimensionless> >
(meter) created WP-Puffer-VL waterstarm 50062401 encrypted
(meter) used meter template WP-Puffer-VL 50062401 waterstarm to match 50062401.M=EFE.V=70.T=06
(meter) started meter 4 (WP-Puffer-VL 50062401 waterstarm) identity mode: id R50062401
(wmbus) parseDLL @0 66
(wmbus) parseELL @10 56
(wmbus) parseNWL @13 53
(wmbus) parseAFL @13 53
(wmbus) parseTPL @13 53
(meter) WP-Puffer-VL: for me? 50062401.M=EFE.V=70.T=06 in 50062401
(meter) WP-Puffer-VL: meter detection did not match the selected driver waterstarm! correct driver is: unknown!
(meter) Not printing this warning again for id: 50062401 mfct: (EFE) Engelmann Sensor, Germany (0x14c5) type: Warm Water (30°C-90°C) meter (0x06) ver: 0x70
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) WP-Puffer-VL: yes for me
(meter) WP-Puffer-VL(4) waterstarm  handling telegram from 50062401.M=EFE.V=70.T=06
(meter) WP-Puffer-VL 50062401.M=EFE.V=70.T=06 "4144C5140124065070068C20E97A260030258AA4FA4EEABB7FF16A3D4D4D195F6905291547F08E0DCAD6D16B6730F3366256ADD4362B3642294B7F6D2B703EC1291E"
(wmbus) parseDLL @0 66
(telegram) DLL L=41 C=44 (from meter SND_NR) M=14c5 (EFE) A=50062401 VER=70 TYPE=06 (Warm Water (30°C-90°C) meter) (driver unknown!) DEV=im871a[00104237] RSSI=-52
(wmbus) parseELL @10 56
(telegram) ELL CI=8c CC=20 (slow_resp sync) ACC=e9
(wmbus) parseNWL @13 53
(wmbus) parseAFL @13 53
(wmbus) parseTPL @13 53
(TPL) num encrypted blocks 3 (48 bytes and remaining unencrypted 0 bytes)
(TPL) AES CBC IV decrypting "8AA4FA4EEABB7FF16A3D4D4D195F6905291547F08E0DCAD6D16B6730F3366256ADD4362B3642294B7F6D2B703EC1291E"
(TPL) IV C5140124065070062626262626262626
(TPL) decrypted  "FA6BECE40C2A45387F32A21DD29F637EB77249CBB3CF97CCEBA8D5A60B3531FA1A3AD29C61CC39F21888B14A303CFD66"
(wmbus) WARNING!! decrypted content failed check, did you use the correct decryption key? Permanently ignoring telegrams from id: 50062401 mfct: (EFE) Engelmann Sensor, Germany (0x14c5) type: Warm Water (30°C-90°C) meter (0x06) ver: 0x70
(telegram) TPL CI=7a ACC=26 STS=00 CFG=2530 (synchronous AES_CBC_IV nb=3 cntn=0 ra=0 hc=0)
(meter) newly created meter (WP-Puffer-VL 50062401 waterstarm) did not handle telegram!

How can i fix this problem?

closed 2025-07-25 by github-actions[bot]

Uzlopak, 2026-05-17:
@UBessle 

Ok, i figured it out. Contact me via E-Mail.

Uzlopak, 2026-05-17:
BTW. Your telegram:

```json
{
    "_": "telegram",
    "consumption_at_set_date_m3": 0,
    "id": "50062401",
    "media": "warm water",
    "meter": "waterstarm",
    "meter_datetime": "2025-04-23 20:17",
    "name": "EFEMeter",
    "set_date": "2024-12-31",
    "status": "OK",
    "timestamp": "1111-11-11T11:11:11Z",
    "total_backwards_at_set_date_m3": 0,
    "total_m3": 297.875
}
```

Uzlopak, 2026-05-18:
Replace the 4 at position 19 with a 0. then it should work.