wmbusmeters/wmbusmeters #1521: Unable to decrypt Tauron amiplus - Apator

konradkreciwilk, 2025-05-01:
I'm not able to decrypt amiplus telegram using key from elicznik. I did 3 times deactivaction/activaction to achive a new key, still the same issues.
Meter looks like is known, added to amiplus driver.

wmbusmeters - lastest 1.19.0

time      : 2025-05-01 19:10:03
model     : Wireless-MBus Mode      : T            Manufacturer: APA         ID        : 86023969
Version   : 2            Device Type: 0x02         Device Type String: Electricity                     Control   : 0x44          Data Length: 143          Data      : 8e4401066939028602027a00008005af14ecffbd1dac6f1beb0d3bd05f514ea7bddc08b4b63a24ac706ef2481735597003d4963bd51f5bf5a50e442f666209f86b4152c88434b9271e55057cc7401b239f4735409f1a7f23a43e1bb54fc543ee8d1a7cb44e4869607db2534e72817d49f768af9531fe5f94162d0b9a397050fea9cd0244347a10f6e041d2d605f019
Integrity : CRC
Control Info: 0x7A       Access number: 0x00       Device Type: 0x00         Configuration Word: 0x0580 Payload Encrypted: 1

https://wmbusmeters.org/analyze/8e4401066939028602027a00008005fa7161c58c410a0b1b3561218e09ebf549ff5faeed54a37ab32c4f4fe0b27a227a834a973e486705570f0941b6da9346060f88a4ead38ed0bcd089b078e9d3df3b9568b33b3567bf650dad41328e6140a4241fa76ba9b9573bc46295d919bbc0f0ed9a1dbb3d15950337c5f58aae9d4bef15c0907b1a2af69fca38f6b49bd7f8:amiplus:696B7B923D31D9DF4A9C7530A713C556

konradkreciwilk, 2025-05-01:
(serial) received ascii ",,,,,,,,,,,,,,,,,,,,,,,<0A>"
(rtl433) checkRTL433Frame "2025-05-01 19:44:04,,,Wireless-MBus,,86023969,,,,CRC,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,8e4401066939028602027a00008005a763ad0cb1420764e43c17dff05272a115bd367b90fac3ede5b42cb17bb2db8436968288ffdb5638c7abad3ab2f9b3d111002f90b2310cee426c65518ba46b649c7f41cdedc0dcdf4fb3b7a34a66f84267e6167a090c91dfeb357a5037caaccbd5fed28ced4986bc90f1dd056737ac9518e27fbc10d03ba6bade26cc60136737,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,2,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,T,2,Electricity,122,0,0,1408,,,,,,,,,APA,68,143,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,<0A>"
(wmbus) parseDLL @0 143
(wmbus) parseELL @10 133
(wmbus) parseNWL @10 133
(wmbus) parseAFL @10 133
(wmbus) parseTPL @10 133
(meter) tauron: for me? 86023969.M=APA.V=02.T=02 in 86023969
(meter) tauron: yes for me
(meter) tauron(1) amiplus  handling telegram from 86023969.M=APA.V=02.T=02
(meter) tauron 86023969.M=APA.V=02.T=02 "8E4401066939028602027A00008005A763AD0CB1420764E43C17DFF05272A115BD367B90FAC3EDE5B42CB17BB2DB8436968288FFDB5638C7ABAD3AB2F9B3D111002F90B2310CEE426C65518BA46B649C7F41CDEDC0DCDF4FB3B7A34A66F84267E6167A090C91DFEB357A5037CAACCBD5FED28CED4986BC90F1DD056737AC9518E27FBC10D03BA6BADE26CC60136737"
(wmbus) parseDLL @0 143
(telegram) DLL L=8e C=44 (from meter SND_NR) M=0601 (APA) A=86023969 VER=02 TYPE=02 (Electricity meter) (driver amiplus) DEV=rtl433[00000001] RSSI=999
(wmbus) parseELL @10 133
(wmbus) parseNWL @10 133
(wmbus) parseAFL @10 133
(wmbus) parseTPL @10 133
(TPL) num encrypted blocks 8 (128 bytes and remaining unencrypted 0 bytes)
(TPL) AES CBC IV decrypting "A763AD0CB1420764E43C17DFF05272A115BD367B90FAC3EDE5B42CB17BB2DB8436968288FFDB5638C7ABAD3AB2F9B3D111002F90B2310CEE426C65518BA46B649C7F41CDEDC0DCDF4FB3B7A34A66F84267E6167A090C91DFEB357A5037CAACCBD5FED28CED4986BC90F1DD056737AC9518E27FBC10D03BA6BADE26CC60136737"
(TPL) IV 01066939028602020000000000000000
(TPL) decrypted  "DCCBB9108918823416ABBA064EA72A69A68164405A1E239350561AF14F6DEBCE18E39B635DAA2C6ADE92CBE7BCE1E67280B0EE4445EB2B9A868F228F6110AF8740020D44795B56B3469D529DC81AF663C9039087ECAF23CA1BF61A8B1E8705BB8DE10F467406837BF681A964335240B938A20CFDE8B67743F89BCBE6AC7E4F9F"
(wmbus) WARNING!! decrypted content failed check, did you use the correct decryption key? Permanently ignoring telegrams from id: 86023969 mfct: (APA) Apator, Poland (0x601) type: Electricity meter (0x02) ver: 0x02
(telegram) TPL CI=7a ACC=00 STS=00 CFG=0580 (AES_CBC_IV nb=8 cntn=0 ra=0 hc=0)
(wmbus) telegram from TODO ignored by all configured meters!

pswiatki, 2025-05-02:
But why? I use a key provided in` eLicznik `and it works perfectly OK.

konradkreciwilk, 2025-05-02:
the issue is related only with wrong generation of key by Tauron (elicznik) ?

pswiatki, 2025-05-02:
Do you want me to compare my` eLicznik `key with the string I put in the` wmbusmeters `config (which works very nicely for me)?

pswiatki, 2025-05-02:
OK, I did just that for you:

[image]

make absolutely sure the` id `and` key `are the same in the config file in `/etc/wmbusmeters.d/` as shown above (` id `==` Adres `), like this:

[image]

konradkreciwilk, 2025-05-02:
I sure 

[image]\

# cat /etc/wmbusmeters.d/tauron
name=tauron
type=amiplus
id=86023969
key=696B7B923D31D9DF4A9C7530A713C556

from  /var/log/wmbusmeters/wmbusmeters.org:

(wmbusmeters) logging started 2025-05-02 16:12:33 _
[2025-05-02_16:12:33] Started config rtlwmbus[00000001] listening on none using fq 868.9M
(wmbusmeters) waiting for telegrams
[2025-05-02_16:12:36] (memory) rss 3772416 peak 3.59 MiB
(wmbusmeters) started meter 1 (tauron 86023969 amiplus) identity mode: id R86023969
[2025-05-02_16:13:04] (wmbus) WARNING!! decrypted content failed check, did you use the correct decryption key? Permanently ignoring telegrams from id: 86023969 mfct: (APA) Apator, Poland (0x601) type: Electricity meter (0x02) ver: 0x02
[2025-05-02_16:13:04] (meter) newly created meter (tauron 86023969 amiplus) did not handle telegram!
(wmbusmeters) shutting down

and also when I run manually wmbusmeters:

# wmbusmeters rtlwmbus:868.9M tauron amiplus 86023969 696B7B923D31D9DF4A9C7530A713C556
Started config rtlwmbus[00000001] listening on none using fq 868.9M

(wmbus) WARNING!! decrypted content failed check, did you use the correct decryption key? Permanently ignoring telegrams from id: 86023969 mfct: (APA) Apator, Poland (0x601) type: Electricity meter (0x02) ver: 0x02
(meter) newly created meter (tauron 86023969 amiplus) did not handle telegram!

pswiatki, 2025-05-02:
Very strange, indeed. I have never faced such problems.

pswiatki, 2025-05-02:
OK, I got something when compared your log to my log:

your log:
```
(telegram) DLL L=8e C=44 (from meter SND_NR) M=0601 (APA) A=86023969 VER=02 TYPE=02 (Electricity meter) (driver amiplus) DEV=rtl433[00000001] RSSI=999
```

My log:

```
(telegram) DLL L=8e C=44 (from meter SND_NR) M=38b3 (NES) A=xxxxxxxx VER=03 TYPE=02 (Electricity meter) (driver unknown!) DEV=rtlwmbus[cmd_0] RSSI=133
```

Not sure why it says `driver unknown` in my case (it is NES, not APA), but decodes properly. What I am interested in is why it states `RSSI=999` whereas in my case it is a modest `RSSI=133`, which is strong, but acceptable. Can it be your RF tuner is massively overloaded and thus causing problems? Then again - you should see CRC errors if this was the case, but I could not spot such issues in your log. Or perhaps APA decryption is somehow incorrect in some (edge?) cases?

konradkreciwilk, 2025-05-03:
In the second post I used wmbusmaster with driver rtl433, when I use rtlwmbus RSSI is reported corretly, both has log about problem with decryption.

using driver rtlwmbus:

`(telegram) DLL L=8e C=44 (from meter SND_NR) M=0601 (APA) A=86023969 VER=02 TYPE=02 (Electricity meter) (driver amiplus) DEV=rtlwmbus[00000001] RSSI=124`

I will try contect with Tauron, maybe meter issue ?

pswiatki, 2025-05-03:
> I will try contect with Tauron, maybe meter issue ?

Do you mean speaking with Tauron regarding the encryption key and its validity? If yes: good idea!
By the way: I like the idea of sipping off (z siorbaniem) telegrams from` rtl_433 `"swiss-army-knife" for ISM protocols. I am interested in doing the same and employing common blocks for a larger number of use cases (than only` wmbus `)

xeroski, 2025-05-05:
Reverse all bytes in the key. Correct key:
56C513A730759C4ADFD9313D927B6B69

pswiatki, 2025-05-06:
But why? Is this a major difference between APA and NES?

xeroski, 2025-05-06:
I don't know how the key is generated. Maybe someone forgot to convert little <-> big endian?

konradkreciwilk, 2025-05-06:
Reversed bytes in the key resolve my issue 
thank you for your involvement @pswiatki @xeroski  !

pswiatki, 2025-05-06:
Nice! Good to know. 
Still: I wonder: a) why this is needed in your case, but not in mine - perhaps someone should talk with Tauron about it,
b) why is that swapping chunks stops at 32-bit size and doesn't go down to smaller parts? Can you explain that, @xeroski ?

pswiatki, 2025-05-06:
OK, it is byte swapping first-last, second-previous to the last and so on, so the last bytes swapped are 8-th and 9-th. Is that correct?

xeroski, 2025-05-06:
https://yoginsavani.com/big-endian-and-little-in-memory/

pswiatki, 2025-05-06:
Yeah, looks like byte-wise always. So, a re-write of the number byte-by-byte reversing the byte order. Got it, thanks!