wmbusmeters/wmbusmeters #1522: unable to read information for wmbusmeter (unknown driver)

Dyowstriker, 2025-05-08:
### Type of request

Question

### OS version

proxmox lxc

### wmbusmeters version

1.19.0

### Your message goes here

Hello , 

When i try to read  information by this command 

`wmbusmeters --debug --t1 rtlwmbus`

i found a result but i have a **driver = unknown**

`(telegram) TPL CI=7a ACC=3d STS=00 CFG=2520 (synchronous AES_CBC_IV nb=2 cntn=0 ra=0 hc=0)
Received telegram from: 23373156
          manufacturer: (MAD) Maddalena, Italy (0x3424)
                  type: Water meter (0x07) encrypted
                   ver: 0x88
                device: rtlwmbus[00000001]
                  rssi: 10 dBm
                driver: unknown!
(wmbus) 000   : 2e length (46 bytes)
(wmbus) 001   : 44 dll-c (from meter SND_NR)
(wmbus) 002   : 2434 dll-mfct (MAD)
(wmbus) 004   : 56313723 dll-id (23373156)
(wmbus) 008   : 88 dll-version
(wmbus) 009   : 07 dll-type (Water meter)
(wmbus) 010   : 7a tpl-ci-field (EN 13757-3 Application Layer (short tplh))
(wmbus) 011   : 3d tpl-acc-field
(wmbus) 012   : 00 tpl-sts-field (OK)
(wmbus) 013   : 2025 tpl-cfg 2520 (synchronous AES_CBC_IV nb=2 cntn=0 ra=0 hc=0 )
(wmbus) 015 CE: D5E681152738CADDC5ED9A8397E40EE5F548384E199356D2AEF8508865A1D66E encrypted`

when i want to use a specific driver  with this command:

`wmbusmeters --logtelegrams --format=json auto:t1 watermeter maddalena 23373156 NOKEY`

this informations appears

`Started auto rtlwmbus[00000001] listening on t1
(meter) watermeter: meter detection did not match the selected driver maddalena! correct driver is: unknown!
(meter) Not printing this warning again for id: 23373156 mfct: (MAD) Maddalena, Italy (0x3424) type: Water meter (0x07) ver: 0x88
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) newly created meter (watermeter 23373156 maddalena) did not handle telegram!`

i have download the specific driver in /etc/wmbusmeters.drivers.d.

I can't find where my mistake comes from and I'm a lost .

sorry  but my english is not very well ! Thanks you for yours help

weetmuts, 2025-05-13:
You need to collect the telegram with --logtelegrams and post it here.

weetmuts, 2025-05-13:
I.e. we need the line with the hex data:
telegram=|................|

Dyowstriker, 2025-05-18:
I'm trying to provide you with this information, but I can't even access those logs anymore :/."

I will get back to you quickly

closed 2025-09-27 by github-actions[bot]