wmbusmeters/wmbusmeters #1523: Alternative driver not accepted (mvt-conflict)

F-HALD, 2025-05-13:
### Type of request

Question

### OS version

Home Assistant 15.2

### wmbusmeters version

1.19.0-5

### Your message goes here

I have run into a problem when trying to implement an alternative driver to a meter, which already has en existing driver. My meter transmits more data points than supported by the existing driver and therefore, I try to implement an alternative using a .xmq-driver.

I do not run auto-driver and point to the specific driver in the meters configuration file.

I then get the following response:

[image]

I cannot remove the mvt-part from the driver - this is also not accepted.
If I try to ignore the mvt-part in my new driver by setting it to something "random" (but still point to this new driver in the meters configuration file) , I get this response:

[image]

How is it possible to have such alternative driver accepted. As the auto-driver is not used, I would not have expected this feature to block the way.

Thanks in advance for any help on this matter.

weetmuts, 2025-05-13:
Hmm, it should be possible to override a built in driver. Where do you put the xmq file? 
Can you provide the xmq file here so that I can test?

F-HALD, 2025-05-13:
Thanks for your prompt reply.

I added it using the side-bar config. option in HomeAssistant:

[image]

So far a pretty empty driver and copied from kampressure just to get the basics up and running first:

`
driver {
    name           = kamheat2
    default_fields = name,id,status,pressure_bar,max_pressure_bar,min_pressure_bar,timestamp
    meter_type     = PressureSensor
    detect {
        mvt = KAM,01,10
    }
}
`

F-HALD, 2025-05-13:
Changing now driver to this:

`driver {
    name           = kamheat2
    default_fields = name,id,timestamp
    meter_type     = HeatMeter
    detect {
        mvt = KAM,35,0C
    }
}`

and changing meters config to this:

`
driver=kamheat2
id=80830005
key=########
name=HeatMeter
`
I get this result:

[image]

F-HALD, 2025-05-15:
> Hmm, it should be possible to override a built in driver. Where do you put the xmq file? Can you provide the xmq file here so that I can test?

If there is more data og information that I can provide to help, just say so.