wmbusmeters/wmbusmeters #1529: Qundis Qheat 5.5 US value from other meter

wachterfr, 2025-05-28:
### Type of request

Existing device improvement

### Meter description

Hello together,

I'm facing a weird issue, sometimes a telegram is received, without a value(total_energy_consumption_kwh), like this telegram:
`3c449344058126683f37720581266893443f040010082004ed39070d263201fd0c11046d19113c3502fd3cc2010dff5f0c0008edff008006130701fffc`

decoded via the wmbusmeters.org website, with the qheat_55_us driver(the same I specified in the meter conf):
`{
    "_":"telegram",
    "media":"heat",
    "meter":"qheat_55_us",
    "name":"",
    "id":"68268105",
    "device_date_time":"2025-05-28 17:25",
    "timestamp":"2025-05-28T18:47:54Z"
}`

But then wmbusmeters sends this JSON, where a value is specified:
`METER_JSON={"_":"telegram","media":"heat","meter":"qheat_55_us","name":"qheat_55_us","id":"68268105","key_date_kwh":0,"total_energy_consumption_kwh":4867,"device_date_time":"2025-05-28 17:25","timestamp":"2025-05-28T18:25:23Z","device":"rtlwmbus[00000001]","rssi_dbm":129}`

The value comes from the before received telegram from another meter. That's a strange behavior I could not understand. Can Anybody help me understand the problem? Thank you in advance!

### Logged telegrams from meter in decrypted format

```shell
3c449344058126683f37720581266893443f040010082004ed39070d263201fd0c11046d19113c3502fd3cc2010dff5f0c0008edff008006130701fffc
```

wachterfr, 2025-08-11:
Thank you for your reply.

1.  The Meter IDs are completely unique
2. The Signal Quality couldn't be the issue I think, the meters are 3 meters away from the antenna
3. I think, that's the only opportunity where the wrong values could come from. Do you have any idea ? 
4. The identityMode ist set to ID so it should be strict
5. The debug log I already turned on. But in the log I can't find any solution

Thank you in advance!

geomax2024, 2025-08-11:
I am leaving on a roadtrip in 4 hours, so have spent 10 minutes brainstorming the queries on openAI and have attached the issue breakdown and possible solutions that can be applied for you to explore the many variants possibly causing your issue. A meter’s unique identifier (ID) is expected to be distinct for each device, but in some WMBus or M-Bus parsing implementations, parts of the ID may be truncated, masked, or incorrectly decoded — for example, if the parser applies a hex-to-string conversion with insufficient width, drops the most significant byte, or mishandles endianness. In such cases, two valid IDs like 1234567 and 1234568 could both be misinterpreted as 123456F due to nibble corruption or character substitution.

When this happens, the system cannot reliably differentiate between the two physical meters. This can cause the data store to mistakenly associate a telegram from one meter with the historical data of the other. If a telegram arrives missing certain fields (e.g., no total_energy_consumption_kwh), the system might “helpfully” fill the gap with the last retained value from what it thinks is the same meter — but is actually a different device. This results in cross-contamination of readings, which can cause significant billing or monitoring errors.... It seems far fetched and unlikely, but in wireless communication protocols stranger things have happened. See attached file for your perusal. Good luck. Hope you can dig a bit deeper with the help of openAI which is a good source of ideas... 

[Driver parsing.docx](https://github.com/user-attachments/files/21722925/Driver.parsing.docx)