wmbusmeters/wmbusmeters #1530: driver "fhkvdataiv" is not found

gorischek, 2025-05-29:
### Type of request

Bug

### HAOS version

2025.5

### wmbusmeters version

1.19.0-5

### Your message goes here

I tried to add a techem head cost allocater, but recommended driver "fhkvdataiv" is not found.
Result from scan:

Received telegram from: xxxxxxxx
          manufacturer: (TCH) Techem Service (0x5068)
                  type: Heat Cost Allocator (0x08) encrypted
                   ver: 0x94
                device: cul
                  rssi: -37 dBm
                driver: fhkvdataiv

gorischek, 2025-08-10:
problem still exists

geomax2024, 2025-08-11:
open the actual fhkvdataiv.cc source file and build a full Home Assistant discovery JSON that matches every exported field : 

----> src/driver_fhkvdataiv.cc

from that .cc file you can extract all the fields that the driver will produce and can make a fhkvdataiv.json MQTT discovery template so HA sees every value.

If you save the relevant field data to a json file in the mqtt_discovery directory and restart the add-on, Home Assistant should automatically discover your Techem HCA with all the fields.

Here’s the JSON code you can drop into
/config/wmbusmeters/etc/mqtt_discovery/fhkvdataiv.json : 

{
  "name": "Techem Heat Cost Allocator",
  "device_class": "energy",
  "state_class": "total_increasing",
  "unit_of_measurement": "HCA",
  "value_template": "{{ value_json.current_consumption_hca }}",
  "icon": "mdi:radiator",
  "json_attributes_template": "{{ value_json | tojson }}",
  "unique_id": "fhkvdataiv_{{ value_json.id }}",
  "availability_topic": "wmbusmeters/status",
  "state_topic": "wmbusmeters/{{ value_json.id }}/state",
  "json_attributes_topic": "wmbusmeters/{{ value_json.id }}/state",
  "device": {
    "identifiers": ["fhkvdataiv_{{ value_json.id }}"],
    "name": "{{ value_json.name | default('Techem HCA') }}",
    "model": "Heat Cost Allocator",
    "manufacturer": "Techem"
  }
}

gorischek, 2025-08-31:
thx a lot!

closed 2025-08-31 by gorischek