wmbusmeters/wmbusmeters #1533: NEW meter: id: 87218402 mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Water meter (0x07) ver: 0x63

brenkem, 2025-05-30:
### Type of request

New device support

### Meter description

The meter Type 173 from "Diehl Meetering" seems not to be supported for now. Please advise how to enable support.

```
# wmbusmeters --logtelegrams --format=json auto:c1 watermeter unknown 87218402 NOKEY
Started auto cul on /dev/ttyUSB0 listening on c1
(meter) watermeter: meter detection did not match the selected driver unknown! correct driver is: unknown!
(meter) Not printing this warning again for id: 87218402 mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Water meter (0x07) ver: 0x63
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) newly created meter (watermeter 87218402 unknown) did not handle telegram!

```

```
# wmbusmeters auto:c1
Started auto cul on /dev/ttyUSB0 listening on c1
No meters configured. Printing id:s of all telegrams heard!
Received telegram from: 87218402
          manufacturer: (DME) DIEHL Metering, Germany (0x11a5)
                  type: Water meter (0x07) encrypted
                   ver: 0x63
                device: cul
                  rssi: -47 dBm
                driver: unknown!
```

### Logged telegrams from meter in decrypted format

```shell
(serial) received binary "6236333434413531313032383432313837363330373335433638433030393939"
(cul) checkCULFrame "b6344A51102842187630735C68C00999"
(cul) no eol found yet, partial frame
(serial) received binary "3030463030324332353341314530313030334435334345343832433735393132"
(cul) checkCULFrame "b6344A51102842187630735C68C0099900F002C253A1E01003D53CE482C75912"
(cul) no eol found yet, partial frame
(serial) received binary "3846313237374143323030343130373130374131333434343342303837423633"
(cul) checkCULFrame "b6344A51102842187630735C68C0099900F002C253A1E01003D53CE482C759128F1277AC2004107107A134443B087B63"
(cul) no eol found yet, partial frame
(serial) received binary "3241373239313842443045323745413436323930313737303030394545393235"
(cul) checkCULFrame "b6344A51102842187630735C68C0099900F002C253A1E01003D53CE482C759128F1277AC2004107107A134443B087B632A72918BD0E27EA462901770009EE925"
(cul) no eol found yet, partial frame
(serial) received binary "3733353141313237303631443941423930313436464243434545373345414536"
(cul) checkCULFrame "b6344A51102842187630735C68C0099900F002C253A1E01003D53CE482C759128F1277AC2004107107A134443B087B632A72918BD0E27EA462901770009EE9257351A127061D9AB90146FBCCEE73EAE6"
(cul) no eol found yet, partial frame
(serial) received binary "4637313939324133373245343239303032393933443642454344333538364131"
(cul) checkCULFrame "b6344A51102842187630735C68C0099900F002C253A1E01003D53CE482C759128F1277AC2004107107A134443B087B632A72918BD0E27EA462901770009EE9257351A127061D9AB90146FBCCEE73EAE6F71992A372E429002993D6BECD3586A1"
(cul) no eol found yet, partial frame
(serial) received binary "3934393433313235343845333939313244344238423438433339383332453539"
(cul) checkCULFrame "b6344A51102842187630735C68C0099900F002C253A1E01003D53CE482C759128F1277AC2004107107A134443B087B632A72918BD0E27EA462901770009EE9257351A127061D9AB90146FBCCEE73EAE6F71992A372E429002993D6BECD3586A19494312548E39912D4B8B48C39832E59"
(cul) no eol found yet, partial frame
(serial) received binary "3633364146383033360D0A"
(cul) checkCULFrame "b6344A51102842187630735C68C0099900F002C253A1E01003D53CE482C759128F1277AC2004107107A134443B087B632A72918BD0E27EA462901770009EE9257351A127061D9AB90146FBCCEE73EAE6F71992A372E429002993D6BECD3586A19494312548E39912D4B8B48C39832E59636AF8036<0D><0A>"
(cul) checkCULFrame RSSI_RAW=54
(cul) checkCULFrame LQI=64
(wmbus) trimming frame A "6344A51102842187630735C68C0099900F002C253A1E01003D53CE482C759128F1277AC2004107107A134443B087B632A72918BD0E27EA462901770009EE9257351A127061D9AB90146FBCCEE73EAE6F71992A372E429002993D6BECD3586A19494312548E39912D4B8B48C39832E59636AF"
(wmbus) ff a dll crc 0-9 35c6 ok
(wmbus) ff a dll crc mid 12-27 2c75 ok
(wmbus) ff a dll crc mid 30-45 b632 ok
(wmbus) ff a dll crc mid 48-63 351a ok
(wmbus) ff a dll crc mid 66-81 2a37 ok
(wmbus) ff a dll crc mid 84-99 8e39 ok
(wmbus) ff a dll crc final 102-111 36af ok
(wmbus) trimming frame A "6344A51102842187630735C68C0099900F002C253A1E01003D53CE482C759128F1277AC2004107107A134443B087B632A72918BD0E27EA462901770009EE9257351A127061D9AB90146FBCCEE73EAE6F71992A372E429002993D6BECD3586A19494312548E39912D4B8B48C39832E59636AF"
(wmbus) trimmed 14 dll crc bytes from frame a and ignored 0 suffix bytes.
(wmbus) trimmed frame A "6344A5110284218763078C0099900F002C253A1E01003D53CE489128F1277AC2004107107A134443B087A72918BD0E27EA462901770009EE9257127061D9AB90146FBCCEE73EAE6F71992E429002993D6BECD3586A1949431254912D4B8B48C39832E596"
(cul) received full T1 frame
(meter) no meter handled  checking 1 templates.
(wmbus) parseDLL @0 100
(wmbus) parseELL @10 90
(wmbus) parseNWL @13 87
(wmbus) parseAFL @13 87
(wmbus) parseTPL @30 70
(wmbus) input to kdf for enc "003A1E01000284218707070707070707"
(wmbus) no key, thus cannot execute kdf.
(meter) watermeter: for me? 87218402.M=DME.V=63.T=07 in 87218402
(meter) watermeter: meter detection did not match the selected driver hydrus! correct driver is: unknown!
(meter) Not printing this warning again for id: 87218402 mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Water meter (0x07) ver: 0x63
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) watermeter: yes for me
(formula) parsing "tariff_counter"
(formula) tokens: FIELD(tariff_counter) 
(formula) handle dventry field tariff_counter into tariff counter
(formula) <DVENTRY tariff_counter>
(formula) parsing "tariff_counter"
(formula) tokens: FIELD(tariff_counter) 
(formula) handle dventry field tariff_counter into tariff counter
(formula) <DVENTRY tariff_counter>
(meter) created watermeter hydrus 87218402 encrypted
(meter) used meter template watermeter 87218402 hydrus to match 87218402.M=DME.V=63.T=07
(meter) started meter 1 (watermeter 87218402 hydrus) identity mode: id R87218402
(wmbus) parseDLL @0 100
(wmbus) parseELL @10 90
(wmbus) parseNWL @13 87
(wmbus) parseAFL @13 87
(wmbus) parseTPL @30 70
(wmbus) input to kdf for enc "003A1E01000284218707070707070707"
(wmbus) no key, thus cannot execute kdf.
(meter) watermeter: for me? 87218402.M=DME.V=63.T=07 in 87218402
(meter) watermeter: meter detection did not match the selected driver hydrus! correct driver is: unknown!
(meter) Not printing this warning again for id: 87218402 mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Water meter (0x07) ver: 0x63
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
(meter) watermeter: yes for me
(meter) watermeter(1) hydrus  handling telegram from 87218402.M=DME.V=63.T=07
(meter) watermeter 87218402.M=DME.V=63.T=07 "6344A5110284218763078C0099900F002C253A1E01003D53CE489128F1277AC2004107107A134443B087A72918BD0E27EA462901770009EE9257127061D9AB90146FBCCEE73EAE6F71992E429002993D6BECD3586A1949431254912D4B8B48C39832E596"
(wmbus) parseDLL @0 100
(telegram) DLL L=63 C=44 (from meter SND_NR) M=11a5 (DME) A=87218402 VER=63 TYPE=07 (Water meter) (driver unknown!) DEV=cul RSSI=-47
(wmbus) parseELL @10 90
(telegram) ELL CI=8c CC=00 (slow_resp) ACC=99
(wmbus) parseNWL @13 87
(wmbus) parseAFL @13 87
(telegram) AFL CI=90
(wmbus) parseTPL @30 70
(wmbus) input to kdf for enc "003A1E01000284218707070707070707"
(wmbus) no key, thus cannot execute kdf.
(meter) newly created meter (watermeter 87218402 hydrus) did not handle telegram!
(wmbus) telegram from TODO ignored by all configured meters!
```

geomax2024, 2025-08-11:
**Use Notepad++ and create a new file.

Paste the following code into it and save the file as diver_DMEtype173.cc  - file type C++ source code. Save the file in the wmbusmeters/src folder and recompile wmbusmeters. driver should show up as DMEtype173, select it and test...**

// driver_DMEtype173.cc
// Driver for Diehl Metering Water Meter Type 173 (example IDs, adjust if needed)

#include "meters_common_implementation.h"

namespace
{
    struct Driver_DMEtype173 : public virtual MeterCommonImplementation {
        Driver_DMEtype173(MeterInfo &mi, DriverInfo &di);
    };

    static bool ok = registerDriver([](DriverInfo& di)
    {
        di.setName("DMEtype173");
        di.setDefaultFields("name,id,status,current_consumption");
        di.setMeterType(MeterType::WaterMeter);
        di.addLinkMode(LinkMode::T1);

		// Adjust these IDs to your actual Diehl Type 173 meter detection IDs:
		// Manufacturer ID (0x11a5 for Diehl Metering), version 0x63,
		// and device types 0x07 and 0x75 (water meter variants)
		di.addDetection(0x11a5, 0x63, 0x07);
		di.addDetection(0x11a5, 0x63, 0x75); // if you want to add other types for same version

        di.setConstructor([](MeterInfo& mi, DriverInfo& di) {
            return std::shared_ptr<Meter>(new Driver_DMEtype173(mi, di));
        });
        return true;
    });

    Driver_DMEtype173::Driver_DMEtype173(MeterInfo &mi, DriverInfo &di) : MeterCommonImplementation(mi, di)
    {
        // Status field (string), often from tpl status field
        addStringFieldWithExtractorAndLookup(
            "status",
            "Meter status from tpl status field.",
            DEFAULT_PRINT_PROPERTIES | PrintProperty::STATUS | PrintProperty::INCLUDE_TPL_STATUS,
            FieldMatcher::build()
                .set(MeasurementType::Instantaneous),
            {
                {
                    {
                        "ERROR_FLAGS",
                        Translate::MapType::BitToString,
                        AlwaysTrigger, MaskBits(0xffff),
                        "OK",
                        {
                            // Add error flag bit descriptions here if known
                        }
                    },
                },
            });

        // Current consumption numeric field, volume in cubic meters
        addNumericFieldWithExtractor(
            "current_consumption",
            "Current water consumption",
            DEFAULT_PRINT_PROPERTIES,
            Quantity::Volume,
            VifScaling::Auto,
            DifSignedness::Unsigned,
            FieldMatcher::build()
                .set(MeasurementType::Instantaneous));

        // Add more fields here as you identify them in telegrams:
        // e.g. total consumption, flow rate, etc.
    }
}

[image]

geomax2024, 2025-08-11:
I hope it works. Please note type correction : {Paste the following code into it and save the file as (driver_DME... not diver_Dme...)driver_DMEtype173.cc - file type C++ source code. Save the file in the wmbusmeters/src folder and recompile wmbusmeters. driver should show up as DMEtype173, select it and test...}

andiweidl, 2025-08-11:
hmm.. it is very strange... can you have a look? i do it.. with your input

> (meter) DerMann-Hydrus: for me? 76111433.M=DME.V=75.T=07 in 76111433
[2025-08-11_15:08:37] (meter) DerMann-Hydrus: meter detection did not match the selected driver DMEtype173! correct driver is: unknown!
(meter) Not printing this warning again for id: 76111433 mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Water meter (0x07) ver: 0x75
[2025-08-11_15:08:37] (meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
[2025-08-11_15:08:37] (meter) to add support for this unknown mfct,media,version combination
(meter) DerMann-Hydrus: yes for me
(meter) DerMann-Hydrus(1) DMEtype173  handling telegram from 76111433.M=DME.V=75.T=07
(meter) DerMann-Hydrus 76111433.M=DME.V=75.T=07 "3E44A5113314117675077AB000300588C9E7A7781C4A84C891CE0B958F9D61EF1E7D510C09F3A0A44DEAA43C03CC7B986D97BAE2364BC907BDC11570DC3CD6"
(wmbus) parseDLL @0 63
(telegram) DLL L=3e C=44 (from meter SND_NR) M=11a5 (DME) A=76111433 VER=75 TYPE=07 (Water meter) (driver unknown!) DEV=cul RSSI=-30
(wmbus) parseELL @10 53
(wmbus) parseNWL @10 53
(wmbus) parseAFL @10 53
(wmbus) parseTPL @10 53
(TPL) num encrypted blocks 3 (48 bytes and remaining unencrypted 0 bytes)
(TPL) AES CBC IV decrypting "88C9E7A7781C4A84C891CE0B958F9D61EF1E7D510C09F3A0A44DEAA43C03CC7B986D97BAE2364BC907BDC11570DC3CD6"
(TPL) IV A511331411767507B0B0B0B0B0B0B0B0
(TPL) decrypted  "2F2F0C159719370084046D3B373F378C041501673600426C1F3C4C157510210004FD170000000002FD7467132F2F2F2F"
(telegram) TPL CI=7a ACC=b0 STS=00 CFG=0530 (AES_CBC_IV nb=3 cntn=0 ra=0 hc=0)
telegram=|3E44A5113314117675077AB00030052F2F_0C159719370084046D3B373F378C041501673600426C1F3C4C157510210004FD170000000002FD7467132F2F2F2F|+6923
(meters) trying field info status(Text)[0]...
(meters) using field info status(Text)[0] to extract 0C15 at offset 19
(meters) trying field info current_consumption(Volume)[1]...
(meters) using field info current_consumption(Volume)[1] to extract 0C15 at offset 19
(meter) Any current_consumption_m3 decoded m3 default m3 value 37199.7 (scale 1)

------------------------------------------------------

> di.setName("DMEtype173");
        di.setDefaultFields("name,id,status,current_consumption");
        di.setMeterType(MeterType::WaterMeter);
        di.addLinkMode(LinkMode::T1);

		// Adjust these IDs to your actual Diehl Type 173 meter detection IDs:
		// Manufacturer ID (0x11a5 for Diehl Metering), version 0x63,
		// and device types 0x07 and 0x75 (water meter variants)
		di.addDetection(0x11a5, 0x75, 0x07);

Thanks

geomax2024, 2025-08-11:
Swap the order of the type and version arguments in addDetection() calls:

di.addDetection(0x11a5, 0x07, 0x63);
di.addDetection(0x11a5, 0x75, 0x63); // if 0x75 is also a type with version 0x63

then recompile and test.