wmbusmeters/wmbusmeters #1536: device not working

Chiill34, 2025-06-02:
### Type of request

Bug

### OS version

Debian

### wmbusmeters version

1.19.0

### Your message goes here

Hello,
Following a restart of my homeassistant VM, I can no longer retrieve a telegram.
I noticed that the usb device had changed so I modified it in the config but it doesn't seem to work.
I've attached screenshots of my usb device, my config and the logs, which are stuck at this stage.

[image]
[image]

Logs:
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nginx.sh
cont-init: info: /etc/cont-init.d/nginx.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun flask (no readiness notification)
services-up: info: copying legacy longrun nginx (no readiness notification)
services-up: info: copying legacy longrun wmbusmeters (no readiness notification)
[13:21:09] INFO: Starting wmbusmeters service.
[13:21:09] INFO: Starting web configuration service.
s6-rc: info: service legacy-services successfully started
[13:21:10] INFO: CONFIG_CONF ...
[13:21:10] INFO: {"device":"rtlwmbus","donotprobe":"/dev/bus/usb/002/002","format":"json","logfile":"/dev/stdout","loglevel":"debug","logtelegrams":false,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[13:21:10] INFO: CONFIG_METERS ...
[13:21:10] INFO: [{"driver":"iperl","id":"00301583","key":"E6C88800DEB868C0D6A84880CE982840","name":"WaterMeter"}]
[13:21:10] INFO: Syncing wmbusmeters configuration ...
[13:21:10] INFO: Registering meters ...
[13:21:10] INFO: Adding meter-0001 ...
[13:21:10] INFO: Generating MQTT configuration ... 
[13:21:10] INFO: Broker 192.168.0.26 will be used.
[13:21:10] INFO: 
MQTT Discovery ...
[13:21:10] INFO:  Adding meter: WaterMeter ...
[13:21:10] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00301583_max_flow_m3h/config
[13:21:10] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00301583_rssi_dbm/config
[13:21:10] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00301583_timestamp/config
[13:21:10] INFO:   Add/update topic: homeassistant/sensor/wmbusmeters/00301583_total_m3/config
[13:21:10] INFO: MQTT Discovery cleanup...
[13:21:10] INFO: Starting Nginx...
2025/06/02 13:21:10 [notice] 94#94: using the "epoll" event method
2025/06/02 13:21:10 [notice] 94#94: nginx/1.26.3
2025/06/02 13:21:10 [notice] 94#94: OS: Linux 5.10.0-21-amd64
2025/06/02 13:21:10 [notice] 94#94: getrlimit(RLIMIT_NOFILE): 1048576:1048576
2025/06/02 13:21:10 [notice] 94#94: start worker processes
2025/06/02 13:21:10 [notice] 94#94: start worker process 249
[13:21:11] INFO: Running wmbusmeters ...
(config) "logtelegrams" "false"
(wmbusmeters) version: 1.19.0
(config) using device: rtlwmbus 
(config) number of meters: 1
(config) template WaterMeter 00301583 iperl
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers
Usage:	 -f frequency_to_tune_to [Hz]
	[-s samplerate (default: 2048000 Hz)]
	[-d device_index or serial (default: 0)]
	[-g gain (default: 0 for auto)]
	[-p ppm_error (default: 0)]
	[-b output_block_size (default: 16 * 16384)]
	[-n number of samples to read (default: 0, infinite)]
	[-S force sync output (default: async)]
	[-D enable direct sampling (default: off)]
	filename (a '-' dumps samples to stdout)
<<<
(shell) return code 1
(serial) EVENT thread interrupted
(shell) 259 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(shell) output: >>><<<
(shell) return code 0
(shell) 260 exited
(main) rtl_wmbus found in path
(main) rtlsdr device 00000001 not currently used.
(serial) EVENT thread interrupted
(main) found specified device (rtlwmbus) that matches detected device (:rtlwmbus[00000001]:0/0)
(main) opening rtlwmbus
Started config rtlwmbus[00000001] listening on none
(shell) exec (capture output) "/usr/bin/rtl_wmbus"
(shell) arg "-h"
(shell) output: >>>/usr/bin/rtl_wmbus: unrecognized option: h
rtl_wmbus: 1.1.0
Usage /usr/bin/rtl_wmbus:
	-o remove DC offset
	-a accelerate (use an inaccurate atan version)
	-r 0 to disable run length algorithm
	-t 0 to disable time2 algorithm
	-d 2 set decimation rate to 2 (defaults to 2 if omitted)
	-v show used algorithm in the output
	-V show version
	-s receive S1 and T1/C1 datagrams simultaneously. rtl_sdr _MUST_ be set to 868.625MHz (-f 868.625M)
	-p [T,S] to disable processing T1/C1 or S1 mode
	-f exit if incoming data stalled for 5 seconds
	-h print this help
<<<
(shell) return code 1
(shell) 265 exited
(rtlwmbus) help /usr/bin/rtl_wmbus: unrecognized option: h
rtl_wmbus: 1.1.0
Usage /usr/bin/rtl_wmbus:
	-o remove DC offset
	-a accelerate (use an inaccurate atan version)
	-r 0 to disable run length algorithm
	-t 0 to disable time2 algorithm
	-d 2 set decimation rate to 2 (defaults to 2 if omitted)
	-v show used algorithm in the output
	-V show version
	-s receive S1 and T1/C1 datagrams simultaneously. rtl_sdr _MUST_ be set to 868.625MHz (-f 868.625M)
	-p [T,S] to disable processing T1/C1 or S1 mode
	-f exit if incoming data stalled for 5 seconds
	-h print this help
(rtlwmbus) using command: ERRFILE=$(mktemp -t wmbusmeters_rtlsdr.XXXXXXXXXXXX) ; echo ERRFILE=$ERRFILE ;  date -Iseconds > $ERRFILE ; tail -f $ERRFILE & /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>>$ERRFILE | /usr/bin/rtl_wmbus -s -f
(rtlwmbus) opening 00000001
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "ERRFILE=$(mktemp -t wmbusmeters_rtlsdr.XXXXXXXXXXXX) ; echo ERRFILE=$ERRFILE ;  date -Iseconds > $ERRFILE ; tail -f $ERRFILE & /usr/bin/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>>$ERRFILE | /usr/bin/rtl_wmbus -s -f"
(serialcmd) opened /bin/sh pid 266 fd 6 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
(serial) waiting for stop
(serial) EVENT thread interrupted
(serial) received ascii "ERRFILE=/tmp/wmbusmeters_rtlsdr.XXXXXXfDoMbI<0A>"
(rtlwmbus) checkRTLWMBusFrame "ERRFILE=/tmp/wmbusmeters_rtlsdr.XXXXXXfDoMbI<0A>"
(rtlwmbus) only text
(serial) received ascii "rtl_wmbus: monitoring flow<0A>"
(rtlwmbus) checkRTLWMBusFrame "rtl_wmbus: monitoring flow<0A>"
(rtlwmbus) only text
(serial) received ascii "2025-06-02T13:21:11+02:00<0A>"
(rtlwmbus) checkRTLWMBusFrame "2025-06-02T13:21:11+02:00<0A>"
(rtlwmbus) only text
(serial) received ascii "Found 1 device(s):<0A>  0:  Realtek, RTL2838UHIDIR, SN: 00000001<0A><0A>Using device 0: Generic RTL2832U OEM<0A>Detached kernel driver<0A>"
(rtlwmbus) checkRTLWMBusFrame "Found 1 device(s):<0A>  0:  Realtek, RTL2838UHIDIR, SN: 00000001<0A><0A>Using device 0: Generic RTL2832U OEM<0A>Detached kernel driver<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "  0:  Realtek, RTL2838UHIDIR, SN: 00000001<0A><0A>Using device 0: Generic RTL2832U OEM<0A>Detached kernel driver<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "<0A>Using device 0: Generic RTL2832U OEM<0A>Detached kernel driver<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Using device 0: Generic RTL2832U OEM<0A>Detached kernel driver<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Detached kernel driver<0A>"
(rtlwmbus) only text
(serial) received ascii "Found Fitipower FC0012 tuner<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) checkRTLWMBusFrame "Found Fitipower FC0012 tuner<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Reading samples in async mode...<0A>"
(rtlwmbus) only text

Can you please help me?
I've been using it for several years and haven't had a bug so far, even after a reboot where I just had to change the donotprobe usb config.

Thx

Chiill34, 2025-07-16:
Anyone can help me?

Its's stay on :

(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Reading samples in async mode...<0A>"
(rtlwmbus) only text

geomax2024, 2025-07-17:
have you tried killing any conflicting processes that might be attaching to your rtl device?:
sudo killall rtl*
then start rtl-wmbus again and test... might be worth a try.

Chiill34, 2025-07-17:
in the logs, i see this.
I dont know why he is trying with ttyACM0 because i have DONOTPROBE=/dev/bus/usb/003/003
In ttyACM0, its my zigbee module....

[17:25:26] INFO: Running wmbusmeters ...
Started auto rtlwmbus[00000001] listening on t1
No meters configured. Printing id:s of all telegrams heard!
[17:28:58] INFO: Starting wmbusmeters service.
[17:28:58] INFO: CONFIG_CONF ...
[17:28:58] INFO: {"device":"auto:t1","donotprobe":"/dev/bus/usb/003/003","format":"json","logfile":"/dev/stdout","loglevel":"debug","logtelegrams":true,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[17:28:58] INFO: CONFIG_METERS ...
[17:28:58] INFO: []
[17:28:58] INFO: Syncing wmbusmeters configuration ...
[17:28:58] INFO: Registering meters ...
[17:28:58] INFO: Generating MQTT configuration ... 
[17:28:58] INFO: MQTT service found, fetching credentials ...
[17:28:58] INFO: Broker core-mosquitto will be used.
[17:28:59] INFO: 
MQTT Discovery ...
[17:28:59] INFO: MQTT Discovery cleanup...
[17:29:00] INFO: Running wmbusmeters ...
(config) "logtelegrams" "true"
(wmbusmeters) version: 1.19.0-10-g1027363
(config) number of meters: 0
(main) device /dev/ttyACM0 not currently used, detect contents...
(serial) check if /dev/ttyACM0 can be accessed
(serial) tty /dev/ttyACM0 can be accessed
(serialtty) device /dev/ttyACM0 is already in use and locked.
(amb8465/3665) could not open tty /dev/ttyACM0 for detection
(serial) EVENT thread interrupted
(serialtty) device /dev/ttyACM0 is already in use and locked.
(im871a) could not open tty /dev/ttyACM0 for detection
(serial) EVENT thread interrupted
(serialtty) device /dev/ttyACM0 is already in use and locked.
(serial) EVENT thread interrupted
(serialtty) device /dev/ttyACM0 is already in use and locked.
(serial) EVENT thread interrupted
(serialtty) device /dev/ttyACM0 is already in use and locked.
(iu891a) could not open tty /dev/ttyACM0 for detection
(main) ignoring /dev/ttyACM0, it does not respond as any of the supported wmbus devices.
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers
Usage:	 -f frequency_to_tune_to [Hz]
	[-s samplerate (default: 2048000 Hz)]
	[-d device_index or serial (default: 0)]
	[-g gain (default: 0 for auto)]
	[-p ppm_error (default: 0)]
	[-b output_block_size (default: 16 * 16384)]
	[-n number of samples to read (default: 0, infinite)]
	[-S force sync output (default: async)]
	[-D enable direct sampling (default: off)]
	filename (a '-' dumps samples to stdout)
<<<
(shell) return code 1
(serial) EVENT thread interrupted
(shell) 557 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(shell) output: >>><<<
(shell) return code 0
(serial) EVENT thread interrupted
(shell) 558 exited
(main) rtl_wmbus found in path
(main) rtlsdr device 00000001 not currently used.
(main) opening rtlwmbus
Started auto rtlwmbus[00000001] listening on t1
[... 50 lines cut ...]
(rtlwmbus) only text
(serial) received ascii "2025-07-17T17:29:00+02:00<0A>"
(rtlwmbus) checkRTLWMBusFrame "2025-07-17T17:29:00+02:00<0A>"
(rtlwmbus) only text
(serial) received ascii "Found 1 device(s):<0A>  0:  Realtek, RTL2838UHIDIR, SN: 00000001<0A><0A>Using device 0: Generic RTL2832U OEM<0A>Detached kernel driver<0A>Found Fitipower FC0012 tuner<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) checkRTLWMBusFrame "Found 1 device(s):<0A>  0:  Realtek, RTL2838UHIDIR, SN: 00000001<0A><0A>Using device 0: Generic RTL2832U OEM<0A>Detached kernel driver<0A>Found Fitipower FC0012 tuner<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "  0:  Realtek, RTL2838UHIDIR, SN: 00000001<0A><0A>Using device 0: Generic RTL2832U OEM<0A>Detached kernel driver<0A>Found Fitipower FC0012 tuner<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "<0A>Using device 0: Generic RTL2832U OEM<0A>Detached kernel driver<0A>Found Fitipower FC0012 tuner<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Using device 0: Generic RTL2832U OEM<0A>Detached kernel driver<0A>Found Fitipower FC0012 tuner<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Detached kernel driver<0A>Found Fitipower FC0012 tuner<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Found Fitipower FC0012 tuner<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Reading samples in async mode...<0A>"
(rtlwmbus) only text