wmbusmeters/wmbusmeters #1546: meterfilesaction=append, but meter file overwritten regardless

pswiatki, 2025-06-28:
### Type of request

Bug

### OS version

Arch Linux  6.15.2-arch1

### wmbusmeters version

wmbusmeters: 1.19.0-14-g917478f 917478f4b758edffdd44ac738236c2c5bf5180aa

### Your message goes here

Even though `/etc/wmbusmeters.conf` contains: 

```
meterfiles=/var/log/wmbusmeters/meter_readings
meterfilesaction=append
meterfilesnaming=name-id
meterfilestimestamp=day
```

I still get the last used meter file overwritten when the machine is rebooted. I wonder if this has anything to do with the occasional updates of the system, or is completely independent (a possible quirk in the `wmbusmeters` itself).
For a long time, I thought it worked OK, but now I have doubts again.

pswiatki, 2025-06-28:
Is the interim solution to use `--meterfilestimestamp=micros` to ensure unique filenames each time the service is restarted?

Side question:

weetmuts, 2025-06-28:
This is odd. The source looks pretty simple, look inte printer.cc
function Printer::printFiles line 160:         

const char *mode = overwrite_ ? "w" : "a";
output = fopen(filename, mode);

Where overwrite_ is false if you use append.

Hmmm.

pswiatki, 2025-06-28:
Maybe under some peculiar circumstances this `overwrite_` has an unexpected value and then it makes a nasty loss of previously accumulated data in the meter file. I guess, I will add a log entry just before this `fopen` so that we know for sure what causes this issue.

weetmuts, 2025-06-30:
It only happens at the restart?

pswiatki, 2025-07-03:
OK, found it in [config.cc, line ~700](https://github.com/wmbusmeters/wmbusmeters/blob/917478f4b758edffdd44ac738236c2c5bf5180aa/src/config.cc#L700)

But what is written in the comment above it is strange:
```
    // --useconfig=/ will work to find /etc/wmbusmeters.conf and /etc/wmbusmeters.d
    // --useconfig=/etc will also work to find /etc/wmbusmeters.conf and /etc/wmbusmeters.d
    // The second one is preferable but for backward compatibility the first one is tested first.
    // If there is no /+etc/wmbusmeters.conf then it will look for /+wmbusmeters.conf
```
How so? No matter if `--useconfig=/etc` or `--useconfig=/` is used - the result would be the same?

closed 2025-10-03 by github-actions[bot]