wmbusmeters/wmbusmeters #1547: Are Itron wMIUrf and hMIUrf devices are supported (not V2) and if so, which USB stick should I get?

michaelmell, 2025-06-28:
### Type of request

Question

### OS version

Plan to use wmbusmeters with Home Assistant OS

### wmbusmeters version

none running yet

### Your message goes here

Hello,

we have these meters and heat meter from Itron / Allmess GmbH in our flat.
I am wondering if they should work with wmbusmeters and if so I would like to request some guidance:

- Water meter: [Itron EquaScan wMIUrf](https://fr.itron.com/o/commerce-media/accounts/-1/attachments/3820843)
-Heat meter: [Itron EquaScan hMIUrf](https://na.itron.com/documents/44647/3824733/EquaScan+hMIU+RF_TS0518_KL.pdf/4eb66cdd-67d5-0b76-4606-fa67c07ef615?version=1.1&t=1711249877864)

Please note that these are **not** the newer V2 models described [here](https://oms-group.org/wp-content/uploads/2024/08/P0406_EquaScan_wMIURF_V2_TS0323_1_.pdf) and [here](https://oms-group.org/wp-content/uploads/2024/08/P0409_EquaScan_hMIU_RF_D_TS0323_1_.pdf) (links in German; I could not find english versions, apologies). 

Note:
I saw that "Itron" is listed in the [README](https://github.com/wmbusmeters/wmbusmeters) under "Supported water meters" and [this file](https://github.com/wmbusmeters/wmbusmeters/blob/master/drivers/src/itronheat.xmq) mentions `Allmess UltraMaXX with ITRON EquaScan hMIU RF Module`, which should be my model (given that it does not explicitly metion V2). So it seems they should be supported.

If they are supported:
What USB stick do you recommend I should get for communicating with them? I saw that the `iu891a` is mentioned often in the README. Would this be a good option? I could get it for around 60 Euros (e.g. [here](https://shop.imst.de/wireless-solutions/lora-products/96/iu891a-xl-prolink-lorawan-usb-adapter-868-mhz?c=7)), which is acceptable.
I plan to operate these on my Raspberry Pi 4 running my Home Assistant instance. Is there anything else I need to consider?

Any input is greatly appreciated. Thank you in advance!

mdewendt, 2025-08-31:
Hello,

the data output of the classic and V2 is the same and both are supported by wmbusmeters.
For details please contact me directly.

Martin

michaelmell, 2025-08-31:
Thank for this information. It  is much appreciated!

closed 2025-08-31 by michaelmell