wmbusmeters/wmbusmeters #1556: Support for HYDRUS 2.0

u-hardy, 2025-07-18:
meter detection could not find driver for id: xxxxxxxx mfct: (DME) DIEHL Metering, Germany (0x11a5) type: Water meter (0x07) ver: 0x61

**Device:** HYDRUS 2.0 from Diehl Metering  
**Manufacturer:** DME (0x11a5)  
**Type:** Water meter (0x07)  
**Version:** 0x61  
**Driver used:** izar  
**Error:** meter detection could not find driver for id: xxxxxxxx

**Telegram:**
telegram=|5344A5112712188761078C800E900F002C256802010073FC5191591F6CE47A68C03107102F2F_0C139216010004FD17000000080A5A0502426CFFFF4C13AAAAAAAA4C933CAAAAAAAA2F2F2F2F2F2F2F2F2F2F2F2F|+44

Please add support for this mfct/media/version combination.

KarstenWehner, 2025-08-03:
Hi u-hardy,
I successfully read a Hydrus 2.0 here in Germany on a Raspberry Pi 4 with a nanoCUL obtained from a shop named schlauhaus.
The data is forwarded to a MQTT broker and displayed in openHAB.
You need the AES key from you water supplier. My supplier ThüringenWasser provided me the key after sending a friendly e-mail.

Here is my configuration (only changed lines):
File: /etc/wmbusmeters.conf
device=/dev/ttyUSB0:t1
shell=/usr/bin/mosquitto_pub -h localhost -t /wmbusmeters/$METER_ID -m

File: /etc/wmbusmeters.d/water
# Config für Wasserzähler Diehl Hydrus, id und key müssen natürlich mit eigenen Werten ersetzt werden
name=water
id=12345678
key=00112233445566778899AABBCCDDEEFF
driver=hydrus

Greetings, Karsten

KarstenWehner, 2025-10-03:
Ich bin bis zum 19.10.2025 abwesend.
Bitte beachten Sie, dass Ihre E-Mail nicht weitergeleitet wird.
In dringenden Fällen wenden Sie sich bitte an ***@***.*** oder
telefonisch an +49 3643 24790 

I am out of office until October 19th 2025.
Please note, that your e-mail is not forwarded.
In urgent cases please contact ***@***.*** or call +49 3643 24790 

Mit freundlichen Grüßen / Kind regards
  Karsten Wehner

Best regards / Mit freundlichen Grüßen

¶

Karsten Wehner

Product Specialist Immunhaematology and Software

 

BECOM Software AG
Kastanienallee 8, D-99428 Weimar
Phone: +49 3643 24790
Fax: +49 3643 247999 
http://www.becom.ag 
Mobile: +49 177 5599655
E-Mail: ***@***.*** 

 

Vorstand:
Dipl.-Ing. Stephan Beyer
Aufsichtsratsvorsitz:
WP/StB Dipl.-BW Gerhard Dürrschnabel
Handelsregister: Jena - HRB 112934 
UStID: DE175410628

¶

Please do not send us Word or Excel files. They will be filtered for
security reasons.
Such files can be send in compressed archives, e.g. ZIP files.

closed 2026-01-02 by github-actions[bot]