wmbusmeters/wmbusmeters #1571: Integra Metering driver

andtsin, 2025-08-16:
### Type of request

Enhancement

### HAOS version

2025.8.1

### wmbusmeters version

1.19.0

### Your message goes here

Congratulation for your work. I have rtl-sdr v4 and i am trying to "read" with hydrus driver an integra metering, although i have decryption key but the output is this:

meter detection did not match the selected driver hydrus! correct driver is: unknown!
(meter) Not printing this warning again for id: 10273624 mfct: (IMT) INTEGRA METERING, Switzerland (0x25b4) type: Water meter (0x07) ver: 0x10
Do you have any idea how to correct it? Thanks in advance

andtsin, 2025-08-16:
(rtlwmbus) checkRTLWMBusFrame "Found Rafael Micro R828D tuner<0A>RTL-SDR Blog V4 Detected<0A>Sampling at 1600000 S/s.<0A>Tuned to 868950000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "RTL-SDR Blog V4 Detected<0A>Sampling at 1600000 S/s.<0A>Tuned to 868950000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Sampling at 1600000 S/s.<0A>Tuned to 868950000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Tuned to 868950000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Reading samples in async mode...<0A>"
(rtlwmbus) only text

And this is my wmbusmeters.conf

geomax2024, 2025-08-17:
An RSSI (Received Signal Strength Indicator) of -80 dBm or better is generally considered good for a stable connection. An RSSI in the range of -80 dBm to -90 dBm is acceptable, but you might start to see occasional dropped packets. Readings below -90 dBm are typically poor and will result in lost telegrams.

A wider bandwidth allows for more data, but can also pick up more noise. wmbusmeters automatically handles the bandwidth for the w-mbus protocol, so you don't need to manually adjust this.

You don't have to change the automatic gain. It's designed to automatically adjust for the best reception. If you were to manually set it, you could lose the ability to receive both strong and weak signals. For most applications, letting the automatic gain control (AGC) work is the best option. 

If the decryption key is incorrect, the telegram will be marked as "decryption failed" in the logs. You will still receive the data packets, but the payload will be an unreadable hex string. If you don't see this specific log entry, it could mean the key is correct or that the meter is not sending encrypted data. For a meter with ID 10273629, you need to use the specific key provided by your utility for that device.

A discone multiband antenna is an omnidirectional antenna. It's great for listening to a wide range of frequencies from all directions. However, for a single, long-range w-mbus link, a directional antenna like the one you ordered (HORIZON-868-HELIUM) would be better, as it's specifically designed for the 868 MHz band and will have a higher gain than a general-purpose discone.