wmbusmeters/wmbusmeters #1572: Erreur fichier lors de la recherche de signal

pelatan, 2025-08-19:
### Type of request

Enhancement

### OS version

raspberry pi4

### wmbusmeters version

1.3.0-45 gdb42776

### Your message goes here

Bonjour,

Quand je lance wmbusmeters, j'ai la recherche qui ne se fait pas et détecte des erreur fichier, quelqu'un peut m'aider  à trouver le problème. Ci-dessous je joins le script :
root@raspberrypi:~# sudo wmbusmeters --debug --t1 rtlwmbus
(drivers) added builtin driver aquastream
(drivers) added builtin driver elf2
(drivers) added builtin driver elster
(drivers) added builtin driver eltako
(drivers) added builtin driver hydrocalm4
(drivers) added builtin driver ime
(drivers) added builtin driver iperl
(drivers) added builtin driver istaheat
(drivers) added builtin driver itronheat
(drivers) added builtin driver izar
(drivers) added builtin driver kampress
(drivers) added builtin driver maddalena
(drivers) added builtin driver relhca
(drivers) added builtin driver sensostar
(drivers) added builtin driver supercal
(drivers) added builtin driver werhlemodwm
(wmbusmeters) version: _
(config) using device: rtlwmbus
(config) number of meters: 0
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_sdr < /dev/null"
(shell) output: >>>rtl_sdr, an I/Q recorder for RTL2832 based DVB-T receivers

Usage:   -f frequency_to_tune_to [Hz]
        [-s samplerate (default: 2048000 Hz)]
        [-d device_index or serial (default: 0)]
        [-g gain (default: 0 for auto)]
        [-p ppm_error (default: 0)]
        [-b output_block_size (default: 16 * 16384)]
        [-n number of samples to read (default: 0, infinite)]
        [-S force sync output (default: async)]
        [-D enable direct sampling (default: off)]
        filename (a '-' dumps samples to stdout)

<<<
(serial) EVENT thread interrupted
(shell) return code 1
(shell) 22003 exited
(main) rtl_srd found in path
(shell) exec (capture output) "/bin/sh"
(shell) arg "-c"
(shell) arg "rtl_wmbus < /dev/null"
(shell) output: >>><<<
(serial) EVENT thread interrupted
(shell) return code 0
(shell) 22005 exited
(main) rtl_wmbus found in path
(main) rtlsdr device 00000001 not currently used.
(main) found specified device (rtlwmbus) that matches detected device (:rtlwmbus[00000001]:0/0)
(main) opening rtlwmbus
Started config rtlwmbus[00000001] listening on t1
[... 56 lines cut ...]
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Found 1 device(s):<0A>  0:  Realtek, RTL2838UHIDIR, SN: 00000001<0A><0A>Using device 0: Generic RTL2832U OEM<0A>Detached kernel driver<0A>Found Rafael Micro R828D tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "  0:  Realtek, RTL2838UHIDIR, SN: 00000001<0A><0A>Using device 0: Generic RTL2832U OEM<0A>Detached kernel driver<0A>Found Rafael Micro R828D tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "<0A>Using device 0: Generic RTL2832U OEM<0A>Detached kernel driver<0A>Found Rafael Micro R828D tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Using device 0: Generic RTL2832U OEM<0A>Detached kernel driver<0A>Found Rafael Micro R828D tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Detached kernel driver<0A>Found Rafael Micro R828D tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Found Rafael Micro R828D tuner<0A>[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "[R82XX] PLL not locked!<0A>Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Sampling at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Reading samples in async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text

pelatan, 2025-08-19:
J'ai essayé d'écrire le script mais rien ne se passe. 

Par contre en le modifiant, j'ai eu quelques informations

root@raspberrypi:/opt/rtl-wmbus# rtl_sdr -f 868.95M -s 1.6e6 - | rtl_wmbus
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R828D tuner
[R82XX] PLL not locked!
Sampling at 1600000 S/s.
Tuned to 868950000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...
Allocating 15 zero-copy buffers

geomax2024, 2025-08-19:
Hope this helps : Courtesy of Gemini 2.5 Flash : 

The main error is [R82XX] PLL not locked!.

This error means that the Phase-Locked Loop (PLL) of the tuner (the Rafael Micro R828D) failed to lock onto the desired frequency of 868.625 MHz. In simple terms, your RTL-SDR device can't "tune in" correctly to the right frequency to receive signals, which is why wmbusmeters isn't detecting any meter telegrams.

Your log shows Detached kernel driver. Normally, the operating system (Raspberry Pi OS) has a generic driver for DVB-T/SDR tuners that can interfere with rtl_sdr or rtl_wmbus. The PLL error can be a symptom of this interference.

You need to "blacklist" the kernel drivers so that the system doesn't automatically load them at boot.

Create a new configuration file:

sudo nano /etc/modprobe.d/rtl-sdr-blacklist.conf

Add the following lines to the file:

blacklist dvb_usb_rtl28xxu
blacklist rtl2832
blacklist rtl2830

Save the file and reboot your Raspberry Pi:

sudo reboot

After the reboot, try running "sudo wmbusmeters --debug --t1 rtlwmbus"    again.

2. USB Port or Power Problem

Insufficient power or an issue with the USB port can prevent the RTL-SDR dongle from functioning correctly, which can lead to the PLL failing to lock.

Use a different USB port on your Raspberry Pi.

If you're using a USB hub, make sure it's an externally powered hub (not a passive one).

If the problem continues, it might indicate a hardware fault with the RTL-SDR dongle itself.

3. Reception or Antenna Issue
While less common, a disconnected or damaged antenna, or a very weak signal, can sometimes cause PLL lock issues.

Make sure the antenna is firmly connected to your RTL-SDR dongle.

Test with another antenna if possible.

Try adjusting the antenna's location to improve reception.

pelatan, 2025-08-20:
J'ai commandé une nouvelle clé, je reviendrai après faire un point.

pelatan, 2025-08-28:
Bonjour,

j'ai reçu ma nouvelle clé et je n'ai plus de problème matériel : 

Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Fitipower FC0012 tuner
Sampling at 1600000 S/s.
Tuned to 868950000 Hz.
Tuner gain set to automatic.
Reading samples in async mode...
Allocating 15 zero-copy buffers

Par contre quand je lance wmbusmeters je rencontre le problème suivant dans l'exécution du programme : 

(serial) EVENT thread interrupted
(rtlwmbus) using command: ERRFILE=$(mktemp -t wmbusmeters_rtlsdr.XXXXXXXXXXXX) ;                                                                                                                                                              echo ERRFILE=$ERRFILE ;  date -Iseconds > $ERRFILE ; tail -f $ERRFILE & /usr/bi                                                                                                                                                             n/rtl_sdr  -d 0 -f 868.625M -s 1.6e6 - 2>>$ERRFILE | /usr/bin/rtl_wmbus -s -f
(rtlwmbus) opening 00000001
(serial) EVENT thread interrupted
(bgshell) exec background "/bin/sh"
(bgshell) arg "-c"
(bgshell) arg "ERRFILE=$(mktemp -t wmbusmeters_rtlsdr.XXXXXXXXXXXX) ; echo ERRFI                                                                                                                                                             LE=$ERRFILE ;  date -Iseconds > $ERRFILE ; tail -f $ERRFILE & /usr/bin/rtl_sdr                                                                                                                                                               -d 0 -f 868.625M -s 1.6e6 - 2>>$ERRFILE | /usr/bin/rtl_wmbus -s -f"
(serialcmd) opened /bin/sh pid 715 fd 6 (rtlwmbus)
(main) regular reset of rtlwmbus  will happen every 82800 seconds
(wmbus) no alarm (expected activity) for rtlwmbus
(serial) registered regular callback HOT_PLUG_DETECTOR(0) every 2 seconds
No meters configured. Printing id:s of all telegrams heard!
(serial) waiting for stop
(serial) received ascii "ERRFILE=/tmp/wmbusmeters_rtlsdr.rituhUZ7u9Ba<0A>rtl_wmb                                                                                                                                                             us: monitoring flow<0A>2025-08-28T17:23:00+01:00<0A>Found 1 device(s):<0A>  0:                                                                                                                                                               Realtek, RTL2838UHIDIR, SN: 00000001<0A><0A>Using device 0: Generic RTL2832U OEM                                                                                                                                                             <0A>Found Fitipower FC0012 tuner<0A>Sampling at 1600000 S/s.<0A>Tuned to 8686250                                                                                                                                                             00 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...<0A>Al                                                                                                                                                             locating 15 zero-copy buffers<0A>"
(rtlwmbus) checkRTLWMBusFrame "ERRFILE=/tmp/wmbusmeters_rtlsdr.rituhUZ7u9Ba<0A>r                                                                                                                                                             tl_wmbus: monitoring flow<0A>2025-08-28T17:23:00+01:00<0A>Found 1 device(s):<0A>                                                                                                                                                               0:  Realtek, RTL2838UHIDIR, SN: 00000001<0A><0A>Using device 0: Generic RTL283                                                                                                                                                             2U OEM<0A>Found Fitipower FC0012 tuner<0A>Sampling at 1600000 S/s.<0A>Tuned to 8                                                                                                                                                             68625000 Hz.<0A>Tuner gain set to automatic.<0A>Reading samples in async mode...                                                                                                                                                             <0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "rtl_wmbus: monitoring flow<0A>2025-08-28T17:23:00                                                                                                                                                             +01:00<0A>Found 1 device(s):<0A>  0:  Realtek, RTL2838UHIDIR, SN: 00000001<0A><0                                                                                                                                                             A>Using device 0: Generic RTL2832U OEM<0A>Found Fitipower FC0012 tuner<0A>Sampli                                                                                                                                                             ng at 1600000 S/s.<0A>Tuned to 868625000 Hz.<0A>Tuner gain set to automatic.<0A>                                                                                                                                                             Reading samples in async mode...<0A>Allocating 15 zero-copy [... cut ...]  n async mode...<0A>Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Reading samples in async mode...<0A>Allocating 15                                                                                                                                                              zero-copy buffers<0A>"
(rtlwmbus) only text
(rtlwmbus) checkRTLWMBusFrame "Allocating 15 zero-copy buffers<0A>"
(rtlwmbus) only text

Pouvez vous m'aider à trouver une solution ? Merci

pelatan, 2025-08-28:
Je viens de faire une mise a jour de wmbusmeters et tout est rentré dans l'ordre il se lance et recherche des signaux.