wmbusmeters/wmbusmeters #1573: Support request: Add driver for Engelmann Sensor (EFE) Heat Meter (MFCT=0x14c5, Type=0x04, Version=0x01) / Supportanfrage: Treiber für Engelmann Sensor (EFE) Wärmezähler hinzufügen (MFCT=0x14c5, Typ=0x04, Version=0x01)

newbie-near-cologne, 2025-08-21:
Hallo zusammen,
ich benutze wmbusmeters mit einem Wärmezähler der Marke Engelmann Sensor (EFE), Modell DE-16-MI004-PTB025. Leider wird dieser Zähler aktuell nicht vom richtigen Treiber erkannt, sondern landet unter „unknown“.

Details zum Gerät:
- Hersteller (MFCT): EFE (0x14c5)
- Gerätetyp: Heat Meter (0x04)
- Version: 0x01
- Meter-ID: 51972777
- 15-stellige ID: 04EFE0151972777

Log-Auszug (Telegramm):
`(meter) Wärmezahler_Fussbodenheizung: meter detection did not match the selected driver microclima! correct driver is: unknown!
(meter) Not printing this warning again for id: 51972777 mfct: (EFE) Engelmann Sensor, Germany (0x14c5) type: Heat meter (0x04) ver: 0x01
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|5E44C5147727975101047A250040252F2F_046D0A302738040600000000C20FFD170000C20F6C0000041300000000043B00000000042B00000000025B1900025F19000261CCFF2F2F2F2F2F2F2F2F2F0F800C00010900627A54001401020A09|+0
`

Ich würde mich freuen, wenn der Treiber für dieses Gerät in das Projekt aufgenommen wird.

Bei Rückfragen oder falls weitere Informationen/Telegramme benötigt werden, stehe ich gern zur Verfügung.

Den AES Schlüssel habe ich und kann die Telegramme auch mit dem microclima Treiber entschlüsseln.
[https://wmbusmeters.org/analyze/5E44C5147727975101047A250040252F2F046D0A302738040600000000C20FFD170000C20F6C0000041300000000043B00000000042B00000000025B1900025F19000261CCFF2F2F2F2F2F2F2F2F2F0F800C00010900627A54001401020A09](url)

Grüße und danke im Voraus

travismarius, 2025-08-22:
Hi @newbie-near-cologne,

Engelmann WMZ werden von vielen als OEM verkauft. Ich habe Engelmann WMZ von Brummerhoop. Die Einbindung hat bei mir funktioniert, indem ich als Treiber "sensostar" angegeben habe und mit der ausgelesenen ID dieses angegeben habe. Ich habe allerdings vorher per Optokopf die AES Verschlüsselung komplett deaktiviert, da ich unterschiedliche Konfigurationen ab Werk drauf hatte.

[image]

mquade, 2025-11-03:
Hallo, ich verwende diese Zähler in allen meinen Mietshäusern. Ich glaube, es ist einer der weitverbreitetsten. Sehr viele OEMs labeln die mit eigenem Namen. Zählerplattform, Brummerhoop, VoluMess, Messwerk... Hab da im Netz schon alles mögliche gesehen. 
Ich bin allerdings Neuling mit wmbusmeters - daher habe ich diesen Thread gefunden. Für das Aufnehmen sage ich +1.
Bei mir ist das Telegram fast identisch, lediglich die Version unterscheidet sich. Aber das würde das Autodetecten vermutlich nicht aus dem Tritt bringen.
```Received telegram from: 52657277
          manufacturer: (EFE) Engelmann Sensor, Germany (0x14c5)
                  type: Heat meter (0x04) encrypted
                   ver: 0x00
                device: iu891a[00202069]
                  rssi: -74 dBm
                driver: sensostar
```
Den AES-Schlüssel habe ich. Sobald ich den Zähler aber in wmbusmeters in HA anlege passiert gar nichts mehr. Aber das gehört hier nicht  hin sondern eher in einen newbe-thread.