wmbusmeters/wmbusmeters #1577: Watermeter APATOR ULTRIMIS Type UL4-190-DN20 Pipersberg

city12001, 2025-08-27:
### Type of request

New device support

### Meter description

Hi, can erveryone help me,

now i get the key to encrypt the telegram, but i get the Error message Driver : Unknown!

How know wich driver i need

Image

[14:05:10] INFO: Starting wmbusmeters service.
[14:05:10] INFO: CONFIG_CONF ...
[14:05:10] INFO: {"device":"auto:c1,t1","donotprobe":"/dev/ttyAMA0","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":true,"shell":"/wmbusmeters/mosquitto_pub.sh "wmbusmeters/$METER_NAME" "$METER_JSON""}
[14:05:10] INFO: CONFIG_METERS ...
[14:05:10] INFO: [{"driver":"unknown","id":"80033850","key":"000F5F67F0179F8CAD776FD**********","name":"Wasserzaehler"}]
[14:05:10] INFO: Syncing wmbusmeters configuration ...
[14:05:10] INFO: Registering meters ...
[14:05:10] INFO: Adding meter-0001 ...
[14:05:10] INFO: Generating MQTT configuration ...
[14:05:10] INFO: MQTT service found, fetching credentials ...
[14:05:10] INFO: Broker core-mosquitto will be used.
[14:05:10] INFO:
MQTT Discovery ...
[14:05:10] INFO: Adding meter: Wasserzaehler ...
[14:05:10] INFO: File /config/wmbusmeters/etc/mqtt_discovery/unknown.json not found.
[14:05:10] INFO: MQTT Discovery cleanup...
[14:05:11] INFO: Running wmbusmeters ...
Started auto iu891a[00202083] on /dev/ttyACM1 listening on t1,c1
(meter) Wasserzaehler: meter detection did not match the selected driver unknown! correct driver is: unknown!
(meter) Not printing this warning again for id: 80033850 mfct: (APA) Apator, Poland (0x601) type: Cold water meter (0x16) ver: 0x20
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination

[image]

[ULTRIMIS-Ausschreibungstexte-wMBus-OMS.pdf](https://github.com/user-attachments/files/22008349/ULTRIMIS-Ausschreibungstexte-wMBus-OMS.pdf)

### Logged telegrams from meter in decrypted format

```shell
telegram=|434401065038038020168C00AB900F002C256CF4AE689FA9B58883F810D07AC4002007102F2F_0413E60E000003FD170C040844130000000004933C000000002F2F2F2F2F|+19
telegram=|434401065038038020168C00AB900F002C257CF4AE68E4F5E2E0BB5DB7587AC5002007102F2F_0413E60E000003FD170C040844130000000004933C000000002F2F2F2F2F|+35
telegram=|434401065038038020168C00AB900F002C258CF4AE68800A4CA2848512927AC6002007102F2F_0413E60E000003FD170C040844130000000004933C000000002F2F2F2F2F|+51
telegram=|434401065038038020168C00AB900F002C259CF4AE685F872836AA2F7B8D7AC7002007102F2F_0413E60E000003FD170C040844130000000004933C000000002F2F2F2F2F|+67
telegram=|434401065038038020168C00AB900F002C25ACF4AE68A5CE34F0CBBC79627AC8002007102F2F_0413E60E000003FD170C040844130000000004933C000000002F2F2F2F2F|+83
telegram=|434401065038038020168C00AB900F002C25BCF4AE685ACD90F164562EF97AC9002007102F2F_0413E60E000003FD170C040844130000000004933C000000002F2F2F2F2F|+99
telegram=|434401065038038020168C00AB900F002C25CCF4AE687F25ABA5B6EB897E7ACA002007102F2F_0413E60E000003FD170C040844130000000004933C000000002F2F2F2F2F|+115
telegram=|434401065038038020168C00AB900F002C25DCF4AE6854A63180D2DBCD6D7ACB002007102F2F_0413E60E000003FD170C040844130000000004933C000000002F2F2F2F2F|+131
telegram=|434401065038038020168C00AB900F002C25ECF4AE6849311E203F7898A87ACC002007102F2F_0413E60E000003FD170C040844130000000004933C000000002F2F2F2F2F|+147
telegram=|434401065038038020168C00AB900F002C25FCF4AE68AE05F63B8D6399E57ACD002007102F2F_0413E60E000003FD170C040844130000000004933C000000002F2F2F2F2F|+163
telegram=|434401065038038020168C00AB900F002C250CF5AE68BF2965B58A798C5E7ACE002007102F2F_0413E60E000003FD170C040844130000000004933C000000002F2F2F2F2F|+179
telegram=|434401065038038020168C00AB900F002C251CF5AE6896EC3A9F82F6AB357ACF002007102F2F_0413E60E000003FD170C040844130000000004933C000000002F2F2F2F2F|+195
telegram=|434401065038038020168C00AB900F002C252CF5AE68D9517D45EB917E627AD0002007102F2F_0413E60E000003FD170C040844130000000004933C000000002F2F2F2F2F|+211
telegram=|434401065038038020168C00AB900F002C253CF5AE6811B4481B9DD78E597AD1002007102F2F_0413E60E000003FD170C040844130000000004933C000000002F2F2F2F2F|+227
telegram=|434401065038038020168C00AB900F002C254CF5AE686820E1E618929C557AD2002007102F2F_0413E60E000003FD170C040844130000000004933C000000002F2F2F2F2F|+243
```

kzajac83, 2025-08-28:
Working good

https://wmbusmeters.org/analyze/434401065038038020168C00AB900F002C256CF4AE689FA9B58883F810D07AC4002007102F2F0413E60E000003FD170C040844130000000004933C000000002F2F2F2F2F

Just replace auto driver into ultrimis

city12001, 2025-08-29:
Hi kzajac83,

thanks for your help and answer, 

but i didn´t foung the right driver, see listing, the software didn´t found ultrimis.json.

Where can i find this driver ??

Thansk for your help in advance

[06:45:06] INFO: Starting wmbusmeters service.
[06:45:06] INFO: CONFIG_CONF ...
[06:45:06] INFO: {"device":"auto:c1,t1","donotprobe":"/dev/ttyAMA0","format":"json","logfile":"/dev/stdout","loglevel":"normal","logtelegrams":true,"shell":"/wmbusmeters/mosquitto_pub.sh \"wmbusmeters/$METER_NAME\" \"$METER_JSON\""}
[06:45:06] INFO: CONFIG_METERS ...
[06:45:06] INFO: [{"driver":"ultrimis","id":"80033850","key":"000F5F67F0179F8CAD776FDFF7869B8B","name":"ultrimis"}]
[06:45:06] INFO: Syncing wmbusmeters configuration ...
[06:45:06] INFO: Registering meters ...
[06:45:06] INFO: Adding meter-0001 ...
[06:45:06] INFO: Generating MQTT configuration ... 
[06:45:06] INFO: Broker 192.168.2.50 will be used.
[06:45:06] INFO: 
MQTT Discovery ...
[06:45:06] INFO:  Adding meter: ultrimis ...
[06:45:06] INFO:   File /config/wmbusmeters/etc/mqtt_discovery/ultrimis.json not found.
[06:45:06] INFO: MQTT Discovery cleanup...
[06:45:07] INFO: Running wmbusmeters ...
Started auto iu891a[00202083] on /dev/ttyACM1 listening on t1,c1
(meter) ultrimis: meter detection did not match the selected driver ultrimis! correct driver is: unknown!
(meter) Not printing this warning again for id: 80033850 mfct: (APA) Apator, Poland (0x601) type: Cold water meter (0x16) ver: 0x20
(meter) please consider opening an issue at https://github.com/wmbusmeters/wmbusmeters/
(meter) to add support for this unknown mfct,media,version combination
telegram=|434401065038038020168C004B900F002C253C30B168224EF12BBBC6B12F7A0A002007102F2F_04134610000003FD170C040844130000000004933C000000002F2F2F2F2F|+7

closed 2025-12-18 by github-actions[bot]