wmbusmeters/wmbusmeters #1579: KEM file processing , with two meter info

mortenx, 2025-09-04:
### Type of request

Enhancement

### HAOS version

2025

### wmbusmeters version

1,19

### Your message goes here

Is it possible for KEM file processing to extract information from 2 meters from one file?

geomax2024, 2025-09-09:
yes a kemfile can contain multiple meters and their keys. It helps to avoid having to add meters to readymanager one by one and allows a batch import of meters into your head-end system... you can use the wmbusmeters/tools/kem-import.py to extract them as long as you have the kemfile password. In my example below I set the password to 12345 when I downloaded my kemfile from mykamstrup.

[image]

BIBOLV, 2025-09-13:
Tnx for file! Fix added.

closed 2025-09-13 by BIBOLV